tinfever/FW-Dyson-BMS #2: Enabling balancing

perigoso, 2022-05-24:
> Cell balancing is not implemented. I know this is ironic, but because the cell balancing resistors aren't installed and Dyson used 1K resistors for the VCELL# connections, even if you shorted out the connections where the cell balancing resistors would go, which most people aren't going to do (and you'd have to cut some very fine traces on the V7 BMS PCBs), the cell balancing would be extremely slow through the 1K resistors. You'd also have to either add #define setting or figure out some way for the firmware to detect if cell balance resistors are installed and then remember it, because you'd need the pack to stay awake on the charger while balancing if applicable, but go to sleep on the charger otherwise. If the pack thinks it is balancing but there are no balance connections, it would stay awake forever. Also I'm burnt out on this project and have worked on my vacuum enough for one lifetime.

you say:
> You'd also have to either add #define setting or figure out some way for the firmware to detect if cell balance resistors are installed and then remember it, because you'd need the pack to stay awake on the charger while balancing if applicable, but go to sleep on the charger otherwise

But I don't this is true, I am yet to look more deeply into the hardware, but you conceivably could stay always awake while on the charger, trying to balance even if physically not possible, and while not on the charger operate normally and go to sleep, the standby power difference would be negligible, at least for a device connected to a power supply, and this way adding balancing by adding the resistors would be innately supported

You also say balancing would be slow because of the 1k VCELL resistors, I don't how the board is wired, but the balancing resistors should have a direct connection to the cell anode, according to the datasheet, so this shouldn't apply?

[image]

anyway, good job with the project, I love this kind of thing.

KerOzenn44, 2023-04-27:
Hello, Does anybody knows if the BMS sold on Aliexpress on Ebay have the balancing function ?

tinfever, 2023-04-27:
> you say:
>
>
> But I don't this is true, I am yet to look more deeply into the hardware, but you conceivably could stay always awake while on the charger, trying to balance even if physically not [...]

Thanks for the interest. Take a look at the schematics in the hardware folder of this repo. You'll see Dyson didn't implement the ISL94208 IC exactly according to the datasheet. On the V7 vacuums they shorted the cell balance pins of the IC to the applicable VCELL pin making balancing impossible unless those traces are cut and rewired. On the V6, if the missing cell balance resistors were shorted out, the balancing current path for Cell 0 in this example would be Cell 0 positive > R7 balancing resistor (shorted) > CB1 pin > VCELL0 pin (via internal balancing FET) > R47 1k resistor > Cell 0 negative. Balancing would be very slow but would probably still work eventually.

On your other point, if I left the microcontroller powered up and the LEDs on permanently when on the charger, it would "micro-cycle" the cells by depleting them enough until the charger kicks in, charges it back up, then disabled charging. I think that might be bad for the cell lifespan. This could probably be worked around though. The only real roadblocks are the cell balancing pin wiring on the V7 making balancing require tricky hardware mods, and the V6 would minor hardware mods (shorting the missing resistors) and would balance slowly. Everything else could be dealt with.

> Hello, Does anybody knows if the BMS sold on Aliexpress on Ebay have the balancing function ?

I'm afraid I don't know. Sorry.

KerOzenn44, 2023-04-28:
If I order one and can see I'll tell you !