tinfever/FW-Dyson-BMS #4: 8 Red flash CHARGE_OC_FLAG

Bartman1001, 2022-05-26:
Hello, I got an 32 red light flash on my SV09 V6 Dyson. I opened battery and manually charged each cell to 4.2V. Then I flashed your custom FW successfully. It worked. But when I try to run with fully charged V6 at Max setting, after 2-3 min later I get an 3 Blue flash as low battery. After a couple min later I tried to run again, and about 1 min alter I get an 3 blue led flash again. But when I try to charge V6 in station, 1 blue, 1 Yellow (unbalance cell I mean) and after those 8 red flash. That mean charging over current. I wait  a minute to try again but every time I got an same 8 red flash. Is this normal. Or a bug on FW. Maybe 2.5ms is too short to read over current situation.

Bartman1001, 2022-05-29:
I solved 8 Red flash issue. I added series resistor to charge port for reducing charge current. When I add 1.2ohm resistor it solved charge over current error. When I measured charge current, I get a max 800mA. Not even Close to 1.4A limit. So I reduce to sense time from 2.5ms to 5ms in isl94208.c file. It solved .now
[FU-Dyson-BMS_V1(5ms).zip](https://github.com/tinfever/FU-Dyson-BMS/files/8793949/FU-Dyson-BMS_V1.5ms.zip)

realphanta, 2022-06-04:
Hey tinfever, I can just join in on thanking you for the effort! I was very happy to find your fw!

I got the same situation as bartman on my V6. 
- 5 year old battery but all cells still in balance
- Removed a cell to measure remaining mAh‘s and brought cell back to same value as the others afterwards
- Solved the lock mode problem with your firmware (it was only 30 red flashes thought, not the 32 everybody is talking about)
- Now after the first usage without any issues I get 8 flashes when connecting the charger

No idea what the problem is yet. I guess I will use your debug tool next to figure out the error in the bms.

realphanta, 2022-06-04:
Update on my issue:
After disconnecting the charger a little longer (sorry for my impatience) a green flashed appeared. The vacuum was reacting to the trigger again and now connecting the charger again does not show any more issues. 
Seems solved for now.

realphanta, 2022-06-14:
Unfortunately the problem is back. 

After one successful charge, I used the V6 for some days until it died using the the powered Fluffy head. From this point I got three blue flashes when triggering (I didn't find an explanation for this code, but assume battery empty) and connecting the charger it went straigfht into 8 red flashes. 

Disconnecting, waiting, triggering in-betwenn, nothing helps this time. 
As per bartmanns observations and following the error code, I tried charging using my lab PSU with reduced voltage. Below 23V works without running into the OC protection. Above 23V I repeatably get 8 red flashes. 
(Sidenote, there could be less cycles of outputting an error. In case the error persists, it will just be triggered again, but when testing, it blocks you for quite some time, every time.)

I still assume this might/should be fixable in software, unfortunately I never did any PIC programming yet, so I'm hesitating.

stanzgy, 2023-01-31:
Thanks for your effort for making this project. I try to use this firmware on my bricked V6 and got same 8 red led flash issue.
Thanks for @Bartman1001's  tips that I just try to modify firmware by increasing OverChargeCurrentTimeout from 2.5ms to 5ms and solved this issue, without any hardware modification.

Here is my modification:
https://github.com/tinfever/FU-Dyson-BMS/blob/5ec961e7164ff81b616d93206fa788e62c97c6d5/firmware/isl94208.c#L37-L44
Change register value on line 44 in isl94208.c from 0b0100110**0** to 0b0100110**1** (take notice of the last bit)

Background:
(screenshots from isl94208 datasheet pdf)
The last two bits in Charge Set is OCCT1 and OCCT0
[image]

The original code set OCCT1,OCCT0->0,0 which leads to 2.5ms OC timeout. Set OCCT0->1 here to increase OC timeout to 5ms. 
[image]

I have tested this modified firmware on my V6 for two weeks with several charge cycles and everything seems ok. But I'm not a hardware expert and not sure whether this modification is dangerous or may damege your device. Try at your own risk.

Mike05dag, 2023-06-19:
Hello, когда я пытаюсь зарядить V7 sv11 на станции, горит синий как обычно и после 20 30 минут этих 12 красных, означает перегрев , почему так происходит,  ведь аккумулятор не был в работе,  не грелся,  собрал ориганальные батарейки и прошивку и просто как обычно поставил на зарядку и вот такая проблема появляется,  что делать подскажите пожалуйста?

Smtblackstone, 2023-11-04:
@stanzgy 8 kırmizi ışık problemini çözdünmü bu düzeltmeden sonra bu sorunu hiç yasadinmi acaba ? Teşekkürler

murtazaaltinkaya, 2024-01-31:
> I solved 8 Red flash issue. I added series resistor to charge port for reducing charge current. When I add 1.2ohm resistor it solved charge over current error. When I measured charge current, I get a max 800mA. Not even Close to 1.4A limit. So I increase to sense time from 2.5ms to 5ms in isl94208.c file. It solved now. [FU-Dyson-BMS_V1(5ms).zip](https://github.com/tinfever/FU-Dyson-BMS/files/8793949/FU-Dyson-BMS_V1.5ms.zip)

 I have a 8 red flash issue same as you... when connect to the charger, 8 red flashing constantly. Before the charger connection, I can connect to the BMS via PİCkit3... but now (after connect to charger), I can not connect to the BMS... PİCkit3 can not connect to the device. how can ı solve the issue ?
when I check the battery charger sprecification, it will give 780mah.. I dont think its a over current issue.. Could you please support me ?

Bartman1001, 2024-01-31:
What you mean by can't connect to the BMS. Is Pickit3 didn't recognize BMS? Did you check your connection pins, are they correct?

murtazaaltinkaya, 2024-01-31:
> What you mean by can't connect to the BMS. Is Pickit3 didn't recognize BMS? Did you check your connection pins, are they correct? 

Yeah pickit3 can not recognize BMS. 
All of the connection is normal. 
You can see the picture as below. 
After the replacement of battery I can connect to the BMS and can load the FW. But after the charger connection, I can not connect o the BMS. Pickit3 say "no device found" as you can see below. 
When I click 1.8V min, V6 battery light flashing (red,green,blue) 
How Can I fix it ? [image][image]
[image]

Bartman1001, 2024-02-01:
The only thing I suggest is a little press to pins as you connect to BMS board. Because at the BMS programing vias could be loose connection.
And also in @thinfever explanation, 
"(Note: I now recommend not connecting the VDD wire at all. The ISL94208 chip seems keen to fail with an externally supply voltage. I'd still suggest waking up the battery pack as describe in step 6 to power the board up for programming. "

murtazaaltinkaya, 2024-02-01:
> The only thing I suggest is a little press to pins as you connect to BMS board. Because at the BMS programing vias could be loose connection.
>
> And also in @thinfever explanation,
>
> "(Note: I now [...]

VDD is not connected on BMC. Pin press performed but no result. I replaced all of the 18650 battery, after the replacement I can connect to the BMS. 
But after connect to the charger, I can not connect to the BMS board. 
-BMS give 8 red flashing when connect to charger.

Bartman1001, 2024-02-01:
Could you give your contact information via email to me please. barismadasoglu@gmail.com

tinfever, 2024-02-02:
It looks like you're doing everything right so I'm not sure why the PICKit can't connect to the microcontroller. Since the BMS board works enough to give you an error code, that means the microcontroller is working at least. My best guess would be an issue in the wires connecting the PICKit to the microcontroller. Maybe there is some conformal coating left in the BMS board holes preventing a connection? Or maybe there is somehow a short between one of the programming pins? Or maybe the jumper wires you're using might be electrically broken somehow?

gregsell, 2026-11-07:
I ran into the same problem (CHARGE_OC_FLAG), though after replacing all cells (new ones: BAK N18650CNP). It kind of  makes sense that new cells sold now have a higher inrush current than the ones sold 10yrs ago. (?)
Changing the current timeout to 5ms as described above did the trick. Thanks for the hint.