tinfever/FW-Dyson-BMS #5: Dyson V10

Kian-Wee, 2022-05-29:
Heya!

Here's the 7s battery pack for the Dyson v10. From what I managed to make out, the battery management ic seems to be the TI bq769300. Not sure what the model number is but the part number seems to be 206340.

[image]
[image]
[image]
[image]
[image]

MakarBondInc, 2022-06-02:
Hi, TI BQ76930, Microchip ATSAMHA0E15A-B or Microchip ATSAMHA1E15A-B. What is the U4 chip?

davidbatoha, 2023-02-07:
Have anyone figured out how to reset this one?

JanekSMC, 2023-03-01:
Man I’d love to get at the firmware of this one! If for no reason other to work out what and how the 2 extra pins work. My gut feel is they’re a 5v logic circuit (so the battery only outputs connected to the Dyson). I’d also love to get a peek at the run commands and limits in the motor circuit IC, at least there look to be easily accessible (though unlabelled) pads on the motor board… just don’t have the balls to blindly stab at it

davidmpye, 2023-08-19:
V10 is sorted :-)

Have a look at https://github.com/davidmpye/V10_Dyson_BMS/wiki

You can flash the pack using a Raspberry Pi as a programmer, or an Atmel-ICE/CMSIS-DAP device.

I've also got a bit of info about what the 2 extra pins do (spoiler alert : a 115200baud bidirectional serial interface between pack and cleaner)

Cheers,

David