tinfever/FW-Dyson-BMS #8: Dyson V8 - Model SV10 - PCB 180207 - Compatible + Tested

lxlxj, 2022-07-30:
[image]
[image]

MrDrGEM, 2022-10-06:
Confirm. Work ok. I make pinout.
[image]

karatektus, 2022-10-18:
can confirm. it works. doesnt support the two other leds which feels kinda clunky, but its alot better than dead

Rocker7837, 2023-02-21:
> Confirm. Work ok. I make pinout. [image]

Could you confirm the pin out? With my multimeter im getting N9 continually to the negative terminal on the battery.

On the photo above is marked as Vss. Dont want to connect them the wring way haha. Thanks

tinfever, 2023-02-22:
@Rocker7837 It is correct that VSS (which means usually means ground) is tied to battery pack ground. The whole BMS board uses that as the ground reference. You can see that in the schematic if you'd like. In short, that pinout is correct.

Rocker7837, 2023-02-22:
Ah ok thanks sorry my eyes confused me with vss an vcc. Thanks for the great work.

Manuel-1981, 2023-11-09:
Hello, 
Does anyone have an original firmware for the SV10, unfortunately I deleted mine.
Thanks

tinfever, 2023-11-09:
> Hello, Does anyone have an original firmware for the SV10, unfortunately I deleted mine. Thanks

I don't think the original Dyson firmware is publicly available. It is read protected on the microcontroller so you can't just read the firmware without bypassing the read protection somehow, which I don't know how to do.

Giovanni-coco, 2025-04-17:
Is it possible to use the same procedure and firmware as v7?