tinfever/FW-Dyson-BMS #9: Dyson V8 - Model SV10 - PCB 180207 trigger pull not waking up

slyker001, 2022-08-06:
When battery in sleep pulling trigger not turning on battery, 3.3v power rising up, no lights or power on terminals present, no reaction on charger. Not goes to sleep again (in standart time).
Powering battery form charger waking it up (when sleep), while idle no problem with running

Flashed not with picKit3
Configuration bits seted by programmer software:
[image]

tinfever, 2022-08-06:
I'm not sure I understand. When you say the 3.3V rail is turning on but there are no LED signals, the output isn't enabled, and then the 3.3V rail doesn't go to sleep, it sounds like the ISL94208 is generating the 3.3V on trigger wakeup signal correctly, but then the PIC isn't responding or operating properly. 

Did the LEDs work before you flashed this firmware? If so, that would indicate the PIC probably isn't faulty.

I would suggest double checking the firmware flashed correctly. I haven't tested using a different programmer so I can't help much with that. I think the configuration bits you show might be correct though.

[image]

What do you mean by "while idle no problem with running"?

slyker001, 2022-08-07:
Before flashing battery worked fine, till 300mv balance error
"while idle no problem with running" means "Solid green - Battery pack is idle. The output isn't enabled and it isn't charging."
if i reset PIC with something battery will work then it goes to sleep and can be waked up by connecting charger and have no problems to reacting on trigger while LEDs show something

And now i messed up something and battery not reacts on charger, will try to reflash. will eeprom data be useful?

slyker001, 2022-08-07:
Sory, tangled up in buttons...

tinfever, 2022-08-07:
I still may not be understanding completely.

> When battery in sleep pulling trigger not turning on battery, 3.3v power rising up, no lights or power on terminals present, no reaction on charger.

It sounds like when the BMS is asleep (no LEDs and 3.3V rail is off) and you pull the trigger with magnet present on the battery, the 3.3V rail turns on, the LEDs stay off, and the output does not turn on. Is that right?

> Powering battery form charger waking it up (when sleep), while idle no problem with running

I think I'm understanding that connecting the charger causes the BMS to wakeup (LEDs light up and 3.3V rail turns on). Then, now that it is awake, if you disconnect the charger and pull the trigger, the output will turn on. Is that right? Does charging work correctly (meaning it charges until full and then stops, then goes to idle, then goes to sleep)?

Could you confirm that you are placing a magnet over the reed switch when you are testing with the trigger? As long as the 3.3V rail is on, the PIC should always be on and at least one LED should always be on. If the 3.3V rail is on but there are no LEDs on, I don't know what could cause that, other than the PIC not operating properly or not programmed properly, or perhaps the V8 BMS board is more different than I thought regarding the LEDs.

The EEPROM data might help. Also, if you are able to upload a short video to YouTube demonstrating the issue, that might help me understand

slyker001, 2022-08-08:
> It sounds like when the BMS is asleep (no LEDs and 3.3V rail is off) and you pull the trigger with magnet present on the battery, the 3.3V rail turns on, the LEDs stay off, and the output does not turn on. Is that right?

Yes
> I think I'm understanding that connecting the charger causes the BMS to wakeup (LEDs light up and 3.3V rail turns on). Then, now that it is awake, if you disconnect the charger and pull the trigger, the output will turn on. Is that right?

Before i added external connector it worked exactly this, after charger also not waking it...
> Does charging work correctly (meaning it charges until full and then stops, then goes to idle, then goes to sleep)?

Yes
interesting detail, if you drop charger voltage a little bit and battery will not charge to 100% it will never goes to idle and sleep :D

> Could you confirm that you are placing a magnet over the reed switch when you are testing with the trigger?

yes, also tried with vacuum, no difference

on video, LED on external connector connected to 3.3v rail (it dim cuz its 12v LED)
programmer in video used to reset PIC, not reflash
https://youtu.be/HU1SUoR8DV0

eeprom data of messed up FW, where i tried enable PWRTE bit
[EEP.zip](https://github.com/tinfever/FU-Dyson-BMS/files/9278937/EEP.zip)

also, prog i'm using not support exactly this PIC, using PIC16LF1829 profile instead. can it be the problem?
and i'm suspicious looking to FOSC configuration bits

tried to power 3.3v rail with external 3.3v PS, battery awaked. can it be slow rising 3.3v rail and PIC not fully awakens? 

tried to fill memory with 0x00 and 0xff, no difference

slyker001, 2022-08-08:
Okay, took oscilloscope, image of rising 3v3 rail
[image]
Soldered another 1K resistor in VCC regulation circuit. not sure it right fix for a problem...
[image]
Not prettiest soldering, but it worked, now battery awakens by pulling trigger and connecting charger. After adding resistor
[image]

Is there configuration bits for startup delay?

tinfever, 2022-08-08:
That video was extremely helpful. I see several things.

- The output is enabled when you press the button, and then is disabled when you release the button. This is correct.
-  When you push the button to enable the output, the LEDs are fading in to show the red-green-blue LED flashes, if you were using a V7 battery. This is correct too.
- When you release the button, the LEDs show 6 blinks, indicated the battery is fully charged. Also correct!
- All of the above LED indicators are running VERY slow. Something is wrong here.
- After the 6 blinks for the battery charging indicator are shown, it takes exactly two minutes for the battery to go to sleep. This should happen in exactly 30 seconds. **I think, for whatever reason, your PIC is running at 1/4 the correct clock speed.**
- When you connect the charger, the 3.3V turns on correctly but the LEDs don't appear to show the cell balance indicator blinks or light up solid when charging. I'm guessing charging is working though from what you've previously said.

I think the main problem is that the PIC is running at 1/4 of the correct clock speed, and this is causing other issues like the PIC not appearing to start when the 3.3V rail is applied, like it should.

The PIC is configured to have the internal oscillator run at 8MHz and then it has the 4x PLL enabled to generate the main 32MHz clock everything runs at. The fact that your PIC seems to be running at exactly 1/4 of the correct speed makes me suspect an issue with that PLL setting. I just compared the datasheet for the PLL setting on the PIC16LF1847 with the PIC16LF1829 your programmer is configured for, but they look the same.

I have no idea why adding that 1K resistor would help, but since it sounds like it did, I'm amazed.

I think the screenshot I sent earlier with the configuration bits was incorrect. I just checked the code and they are configured this way:

```
#pragma config FOSC = INTOSC    // Oscillator Selection->INTOSC oscillator: I/O function on CLKIN pin
#pragma config WDTE = ON    // Watchdog Timer Enable->WDT enabled
#pragma config PWRTE = OFF    // Power-up Timer Enable->PWRT disabled
#pragma config MCLRE = ON    // MCLR Pin Function Select->MCLR/VPP pin function is MCLR
#pragma config CP = OFF    // Flash Program Memory Code Protection->Program memory code protection is disabled
#pragma config CPD = OFF    // Data Memory Code Protection->Data memory code protection is disabled
#pragma config BOREN = ON    // Brown-out Reset Enable->Brown-out Reset enabled
#pragma config CLKOUTEN = OFF    // Clock Out Enable->CLKOUT function is disabled. I/O or oscillator function on the CLKOUT pin
#pragma config IESO = ON    // Internal/External Switchover->Internal/External Switchover mode is enabled
#pragma config FCMEN = ON    // Fail-Safe Clock Monitor Enable->Fail-Safe Clock Monitor is enabled

// CONFIG2
#pragma config WRT = OFF    // Flash Memory Self-Write Protection->Write protection off
#pragma config PLLEN = ON    // PLL Enable->4x PLL enabled
#pragma config STVREN = ON    // Stack Overflow/Underflow Reset Enable->Stack Overflow or Underflow will cause a Reset
#pragma config BORV = LO    // Brown-out Reset Voltage Selection->Brown-out Reset Voltage (Vbor), low trip point selected.
#pragma config LVP = ON    // Low-Voltage Programming Enable->Low-voltage programming enabled
```

Comparing with your bits, it looks like your FOSC is set differently, and PWRTE should be disabled.

I also looked at your EEPROM data and I think all I see is a few "ISL_INT_OVERTEMP_PICREAD" errors but those wouldn't be causing the main issue you are having.

BTW, writing to the EEPROM while using my firmware will not change anything. I only use the EEPROM for fault logging. The firmware effectively never reads from the EEPROM, other than to increment the total runtime counter.

slyker001, 2022-08-09:
Good. Now set configuration to
[image]
Indication now work much faster and it fixed red LED, i don't see it it before (on this FW)
and main - it fixed startup problem, I'm desoldered additional resistor and it trigger pull wakes it up with no problem

i'm thinking placing required conf.bits to "How to install it" or "FAQ" section of readme will be good

video of working battery after setting new configuration bits https://youtu.be/Ld364PlXZIg

and feature suggestion: i saw on another custom FW (closed source) some "service mode" where batt sends debug information via LED, in their case to arduino https://disk.yandex.ru/i/feUHRnRd89j0Tg

closed 2022-08-09 by slyker001

slyker001, 2022-08-09:
Tested with vacuum, startup problem definitely gone
Thank you for this project
And thank you for support

tinfever, 2022-08-09:
I'm glad it's working now for you! 

When I come back to this project, I'll add some info on programming by including the configuration bits and also mentioning your success using a programmer configured for the PIC16LF1829.

That video you showed with the service mode sending battery debug information via the LED is very cool! :D Although you probably know it wouldn't have helped with your problem since the clock on your PIC was running too slow. I think the number of people that would build something like that Arduino debug tool would be extremely small, so it may not warrant the development time. However, it's still very cool and I could sell the tool (with the hardware/firmware all being open source too) and I could see some people buying it just to support the work that went in to this project. Definitely an interesting suggestion. Thanks!