tinfever/FW-Dyson-BMS #10: Dyson v8 SV10 - PCB 233725-01/04

yarix, 2022-08-07:
[image]

* in my case, all 6 cells are around 4v (4.049, 4.054, 4.060, 4.059, 4.091, 4.052)
* i do get 24.36v from the battery pack (between the 2 large metal plate)
* when i add a magnet and click the micro switch:
  * all 3 blue LEDs turn on, and i get 24.36v at the board output.
  * BUT, after 2 second 2 LEDs are turning off, and after another second, the last LED turns off and the voltage on the board output drops to zero (0).

1. have someone noticed similar behavior? 
2. is there compatible FW for this board?

Even if i don't get solution, thank you for this project! 🙏 - very interesting!

[image]

tinfever, 2022-08-07:
Hi. 

Here is a brief excerpt from a post I made on the EEVBlog forum regarding V8 compatibility:

> I'm not sure about V8 models. I just looked at some of the V8/SV10/PCB 180207 (I hate Dyson naming) photos posted dvd4me in the other thread (https://www.eevblog.com/forum/reviews/dyson-v7-trigger-cordless-vacuum-teardown-of-battery-pack/msg4035538/#msg4035538) as well as the schematics from someone else who reverse engineered that PCB (https://github.com/dr-mark-roberts/open-dyson-battery/blob/master/hw/docs/original_bms_v8_180207.svg). If you have that same PCB model, I think it should functionally work, however the LEDs will be nonsensical because the V6/V7 PCBs I've been working on have a single RGB LED, but on that PCB they replaced it with three blue LEDs and one red LED. It translates like this:
> 
> V7 LED color | V8 LED
> Red              | Red
> Green           | Blue LED 1
> Blue             | Blue LED 3
> n/a               | Blue LED 2
> 
> So other than the LEDs being bananas, I think it should work functionally but I can't guarantee it of course. The thermistor connections and pull-up resistor appear the same as the V7, and the trigger/charger detect circuitry also looks the same, so I don't see why it wouldn't work. 

So I think it should work on a V8 and someone reported back saying that it worked for them. I just realized that you are using PCB 233725 which is one that I haven't seen before. Other V8 users were reporting success with PCB 180207, which is different of course.

> BUT, after 2 second 2 LEDs are turning off, and after another second, the last LED turns off and the voltage on the board output drops to zero (0).

Just to confirm, are you running my firmware when this happens?

Also to double check, when you say you "click the micro switch", are you referring to the actual microswitch with the wires soldering to the PCB, or to the button on the battery casing that actuates the microswitch. It's important to note that when you pull the trigger on the vacuum, you are actually pushing the button on the battery casing, when the *releases* the microswitch to turn the output on. Because of this, I usually try to keep the microswitch inside the applicable half of the battery casing so I can use the plastic button as normal. Looking at your photos it looks like you might already be doing this though.

What happens when you press and hold the microswitch? What happens with you leave the microswitch unpressed?

The LED pattern you describe sounds most like the red-green-blue LED flashes that run when the output is enabled, but of course the V8 BMS boards have a different LED arrangement so the colors are different. When you say after a second the last LED turns off, is it literally a second or is it much less than that? The red-green-blue LED flash sequence is fast and all occurs in less than one second. I'm not actually sure what would be causing that LED pattern if it isn't that.

yarix, 2022-08-08:
>PCB 233725 which is one that I haven't seen before

that's why i posted 😃, i saw some threads confirming PCB 180207

> Just to confirm, are you running my firmware when this happens?

No, i'm still on the original firmware. 

> when you say you "click the micro switch", are you referring to ...

Yes, it's the microswitch connected with wires to the PCB, and i keep the small plastic & spring in place. so, as you said, when i click the switch, it actually release / open the circuit. (sorry for the confusion).

>What happens when you press and hold the microswitch?

when i pressed the switch (meaning opening/releasing the microswitch) - i get 24v power at the output for ~1-2 seconds. the 3 blue LED turns on, then 2 LEDS turn off, and finally the last LED turn off.

>When you say after a second the last LED turns off, is it literally a second or is it much less than that? 

might be less then a second, i don't have a good way to monitor it exactly.

tinfever, 2022-08-08:
Ah! You are on the original firmware. That helps. 

I think your battery might actually be functioning properly. The original firmware automatically disables the output if it does not detect a current draw of at least a few amps (if I recall correctly) within a few seconds. That sounds like exactly what you are seeing. If you apply a 2A+ load to the output terminals, and then try to press the button to enable the output, I think the output might stay enabled since it will see the load it is looking for.

What was the original issue that brought you here?

yarix, 2022-09-08:
sorry for the delay in response.
the original issue was a dying battery after few minutes of working in slow speed. And after trying 2 other Aliexpers crap, I discovered this FW and thought to give the original battery new life.
anyway I just bought a new original battery...

thank you for the knowledge you've shared here!

closed 2022-09-08 by yarix