tinfever/FW-Dyson-BMS #11: Dyson V10 Battery pack PCB

black-raison-detre, 2022-08-29:
[image]

Here is a high resolution picture of a dyson v10 batterypack PCB.
I have identified following main ICs
Microchip Atmel mcu:
https://www.microchip.com/en-us/product/ATSAMD20E18
TI battery monitor:
https://www.ti.com/product/BQ76930

I will try to map out the header pin soon

black-raison-detre, 2022-08-29:
PCB number: 220288-01/09

black-raison-detre, 2022-08-29:
[image]
I have map out the pinout to the ATSAMD20 MCU package QFN32
Top to bottom
RESET
PA31  SWDIO
PA30  SWCLK
PA00
PA11
PA10
GND
Vdd

black-raison-detre, 2022-08-29:
So this battery protection board use a different microcontroller. A new firmware will be required for it to work. And SAM D20 family of microcontroller have protection against firmware dumping. Which mean reverse engineering the original firmware will be hard. Unless someone managed to crack the protection or the whole PCB layout is mapped out and a new firmware is written from scratch.

tinfever, 2022-08-29:
Thanks for posting what you found!

Your conclusion makes sense. While maybe the high-level parts of my firmware could be adapted to this, my code is pretty far from clean so I'm not even sure that would be a good idea. Either way, the whole board would probably have to be reverse engineered, although I'd be surprised if the architecture was that different from the V6/V7/V8 BMS boards.

black-raison-detre, 2022-08-31:
Base on my experience with two dyson V10.  The first battery become unusable after about a year and a half of usage. It is not completely dead but it only able to sustain medium power for a few minutes. Max power is no go as it will only last few second. I disassemble the pack and test the individual cells. These are Molicel P26, rated at 2600mah. They still have 2200mah of usable capacity but the internal resistance have raised quite a bit. The battery management board just stop the discharge when pack voltage drop below a certain level and signal the battery is empty. But in reality it is just the short term voltage drop. The cells are still over 4V each!! So what is the point of such a complex battery management circuitry? For a high current battery pack discharge voltage is not a reliable indicator of battery charge level. In a design like this even a slight aging of the cells will make a massive drop of usable capacity.

tinfever, 2022-08-31:
The max power mode on the V7 vacuums uses substantially more power (17A) vs the normal mode (3A). It sounds like you're finding the cells are actually degrading, rather than just becoming imbalanced. I'm not sure how much firmware could do to help with that. It would be an interesting idea to have firmware that could monitor measure the current draw and monitor the cell voltage droop under it to calculate the ESR and take it into account when deciding where the low voltage cutoff should be. I'm not enough of a battery expert to know, when a cell datasheet specs for a 2.7V cutoff, if it matters whether that is under load or not. Although if the cells are already degraded, I doubt you could wring a ton more out of them, although with care there probably isn't much risk in trying.

I'm surprised how quickly you are finding the cells to fail though. It makes me wonder if either the cells you've been getting are poor quality, if the BMS is doing something to inadvertently degrade them over time while the vacuum sits on the charger (like constantly charging and slowly discharging them via a LED or something so the cells get cycle wear just sitting on the charger), or maybe you just use your vacuum in max power mode or just use it in general more frequently that other people do.

The V10 batteries should have a coulomb counter for more accurate state of charge measurements, since it can, I think, monitor how much energy is added and removed from the cells to make a much more informed calculation of the cells state of charge. I expect there is still a low voltage cutout overriding it anyways though.

tinfever, 2022-08-31:
I just found some more information you might find interesting: https://lygte-info.dk/info/BatteryChargePercent%20UK.html
Particularly the graph under "The difference between loaded and unload voltage is greater."

It's remarkable how it shows how the voltage of a cell under load starts to decrease dramatically near depletion, while the voltage measured at no load still appears fine. Near the end a cell with a no-load voltage of 3.65V-ish drops rapidly down to 2.5V under a 3A load. It looks like lower the cutout voltage limit, wouldn't really provide much more battery runtime. The slope there makes it look like the voltage would just keep plummeting.

Although, all that data is for a healthy cell nearly discharged, which may be different from a worn cell with high ESR that is nearly fully charged.

black-raison-detre, 2022-08-31:
I do use high power mode a lot on my first vacuum. That could have a significant impact on the cells longevity. V10 is rated at 525w max on the label. So 7 cells at 29.4V fully charged would result in a current of 17.85A. Current will go up as the battery discharge as dyson vacuum do maintain their suction power all the way till battery depletion. So yes a lot of stress on the battery.
My second vacuum still going strong as I am not using the max power mode as frequent as before. Max power runtime have reduced but medium power mode is still ok.

jayshin, 2023-02-05:
My one is blinking 20 times and not being charged, so I want to replace the bad cells. Is there any way to just reset the BMS?

davidmpye, 2023-05-17:
I've started work on this BMS, first reverse engineering it, then writing replacement f/w for it...

First step is to snoop the i2c traffic between the mcu and the bms IC, so I can work out basic settings etc. 
The other complication in the v10 battery is there is a 2-wire connection to the vacuum as well, not sure what it's for or what travels along it yet, so that'll need figuring out too...

davidmpye, 2023-05-17:
Ah, just figured out the first problem. Spent a while looking at the I2C traffic to see if the status is 'tripped' there when a pack fails, but actually, this MCU has a flash area and EEPROM emulation, so I am almost certain it will store the 'failed' status there.  
There's no way to read this externally and reset it.

The only way to reanimate dead packs is with a whole new FW, which is possible, but will take me a while....!

davidmpye, 2023-08-19:
OK, V10 solution is now ready :-)

Have a look at https://github.com/davidmpye/V10_Dyson_BMS/wiki

You can flash the pack using a Raspberry Pi as a programmer, or an Atmel-ICE/CMSIS-DAP device.

Cheers,

David

jayshin, 2023-08-20:
Great work, David!

What is the minimum requirement for Raspberry Pi? Is it okay with Raspberry Pi Zero?

Thank you!

davidmpye, 2023-08-20:
Hi,

Any 'real' pi - zero/1/2/3/4 etc should work just fine.

The MCU pi Pico series would not.

Cheers

David

jayshin, 2023-08-20:
Hi David,

Thank you for the reply. I just ordered pi 4. Hopefully, I can try it tomorrow.

Regards,
Jay

davidmpye, 2023-08-20:
How exciting - let me know how you get on :-)

David

black-raison-detre, 2023-08-21:
Tanks, I will give it a try. Just hope I haven't fried the first PCB, I think I had overvolt the mcu a bit.