tinfever/FW-Dyson-BMS #12: Dyson V8 Battery 180207 Overcurrent

karatektus, 2022-10-20:
Hey, im running your firmware on a dyson v8 battery and aside from the fact that it doesnt support the 3 Leds it works so far. 
However the startup overcurrent protection kicks in relatively often. (10 red blinks) and i have to wait for a minute for the error to clear. Is this a configuration issue in the release hex?

tinfever, 2022-10-21:
There isn't a configuration issue, at least as far I as I know right now.

Your issue does sound suspiciously similar to someone on the EEVblog forum that was *always* getting the 10 blink error code for short circuit fault on a V8 PCB 180207. It's possible the way I configured it is right on the edge of working on the V8 vacuums, and since I don't have a V8 I never ran in to that issue.

The previous person was getting the short circuit protection code even when the battery wasn't inside the vacuum, so it made most sense that it was probably a hardware fault there.

A few questions to get some more info:

It sounds like this issue is intermittent for you, is that correct? If so, how often would you roughly estimate you get the SCP error code instead of having the vacuum turn on as you expect. 50% of the time? 10%? 90%?

Does charging the battery pack seem to work normally?

If the issue is relatively easy to reproduce, could you see if the issue still occurs without the battery pack installed in the vacuum?

What was the original fault/reason that prompted you to install the firmware on your battery pack?

karatektus, 2022-10-21:
Hey, 

-no it did not occur for me without the pack being installed. i would say it mostly happens when the vacuum comes fresh from the charger (but not always)

-about 10% of the time?

-charging seems to work fine

- the original issue was 3 red blinks (one cell was >0.3V below the others, so the battery was locked) I was quite angry @dyson after i charged that cell up and the bms was still locked :D

tinfever, 2022-10-27:
Hi,

Sorry for the delay. It sounds like the firmware is right on the edge of tripping the SCP protection on startup. I'm not sure why the startup current would be higher on a V8 but I don't have one to test.

I've attached a modified firmware version with a higher SCP limit setting. It is now set to 325A for 190us, instead of the 175A for 190us previously. Could you give that a try and see if it makes any difference?

You'll need to change the file extension back to .hex. GitHub wouldn't let me upload it as a .hex file.

[FU-Dyson-BMS V1_325A SCP Mod.txt](https://github.com/tinfever/FU-Dyson-BMS/files/9875429/FU-Dyson-BMS.V1_325A.SCP.Mod.txt)

karatektus, 2022-11-02:
Hey,

no worries at all! I installed it yesterday and used the vacuum about 5 times since. Always charging it in between. So far it works.
I'll  update this issue if I get any errors here, but if I dont within the next week I guess this is solved.

tinfever, 2022-11-02:
Great! Thanks for reporting back!

mattkellaway, 2023-02-04:
Yeah I got overcurrent on a 180207 just about every time I started it - the batteries are a bit degraded but still holding 4.195v.
I could start it about 5% of the time, no errors without the vac attached, no errors with a 5A resistive dummy load on the battery.

with this variant it starts fine and runs fine, I've cycled the pack a couple of times since and it's great!

Thanks for your work tinfever!

quy54kg, 2023-08-14:
> CHÀO,
> 
> Xin lỗi về sự chậm trễ. Có vẻ như phần sụn đang ở ngay trên bờ vực của việc ngắt lớp bảo vệ SCP khi khởi động. Tôi không chắc tại sao dòng khởi động lại cao hơn trên động cơ V8 nhưng tôi không có cái nào để kiểm tra.
> 
> Tôi đã đính kèm phiên bản phần sụn đã sửa đổi với cài đặt giới hạn SCP cao hơn. Nó hiện được đặt thành 325A cho 190us, thay vì 175A cho 190us trước đây. Bạn có thể thử điều đó và xem liệu nó có tạo ra sự khác biệt nào không?
> 
> Bạn sẽ cần thay đổi phần mở rộng tệp trở lại .hex. GitHub sẽ không cho phép tôi tải nó lên dưới dạng tệp .hex.
> 
> [FU-Dyson-BMS V1_325A SCP Mod.txt](https://github.com/tinfever/FU-Dyson-BMS/files/9875429/FU-Dyson-BMS.V1_325A.SCP.Mod.txt)

thanks for your contribution, the firmware has fixed the overcurrent error on my dyson v8, but there is still 1 error 12 flashing red, is there any firmware version that can fix the 2 errors at the same time?

turenabo, 2024-02-21:
> There isn't a configuration issue, at least as far I as I know right now.
> 
> Your issue does sound suspiciously similar to someone on the EEVblog forum that was _always_ getting the 10 blink error code for short circuit fault on a V8 PCB 180207. It's possible the way I configured it is right on the edge of working on the V8 vacuums, and since I don't have a V8 I never ran in to that issue.
> 
> https://www.eevblog.com/forum/projects/fu-dyson-bms-an-(unofficial)-firmware-upgrade-for-dyson-v6v7-vacuum-bms/msg4410211/#msg4410211
> 
> The previous person was getting the short circuit protection code even when the battery wasn't inside the vacuum, so it made most sense that it was probably a hardware fault there.
> 
> A few questions to get some more info:
> 
> It sounds like this issue is intermittent for you, is that correct? If so, how often would you roughly estimate you get the SCP error code instead of having the vacuum turn on as you expect. 50% of the time? 10%? 90%?
> 
> Does charging the battery pack seem to work normally?
> 
> If the issue is relatively easy to reproduce, could you see if the issue still occurs without the battery pack installed in the vacuum?
> 
> What was the original fault/reason that prompted you to install the firmware on your battery pack?

I see a V8 schematic in your another project, the shunt is 0.001, but I see a V8 board picture it's 0.002,  is it possible there are 2 verions of boards?
[image]
[image]

naneuri, 2024-08-06:
> Hi,
>
> Sorry for the delay. It sounds like the firmware is right on the edge of tripping the SCP protection on startup. I'm not sure why the startup current would be higher on a V8 but I don't have [...]

Thanks for this modified firmware. Could you please let me know which values ​​I need to change below for this? I understand that changing the 2nd and 3rd values ​​of 05H addr to changes Vscd to 0.65A according to the ISL94028 datrasheet, but looking at the source of the existing firmware, it is set to 2 as shown below, which is confusing.
According to the datasheet,  the bit set as 00=0.2V, 01=0.35V, 10=0.65V, 11=1.2V, I guess, but looking at the source, it already has a value of 2, but my understanding is that it for 0.65A.

```
    .OC_DISCHARGE_TIMEOUT_2bits = {0x05, 0, 2},   //0x05 Discharge Set Register, value requires 2 bits     
    .SC_DISCHARGE_THRESH_2bits = {0x05, 2, 2},
```

[image]