tinfever/FW-Dyson-BMS #14: Dyson V7 - PCB 228499 - Compatible + Tested

NemoLee, 2022-10-25:
Great Job! I have tested on V7 -with PCB No. 228499-01/03 it works well

tinfever, 2022-10-27:
Great! Thank you for reporting your results!

tinfever, 2022-10-27:
I added the PCB 228499 to the list of supported models. Could you confirm the model number of your vacuum/battery? It's usually something like SV##, like SV09 for example.

slimlesaurien, 2023-03-21:
Hello sir, I managed to flash the bms (228499) with the tutorial on github (FU-Dyson-BMS), the led reacted correctly. The cells are correctly balanced and the voltage is good.
The only problem I encounter, when I use the vacuum cleaner in boost mode the battery only takes 10 seconds, the vacuum cleaner stops.
I put it back changed everything is correct and I always try again the same.
Can you help me I am new to programming.
Cordially.

NemoLee, 2023-03-22:
> Hello sir, I managed to flash the bms (228499) with the tutorial on github (FU-Dyson-BMS), the led reacted correctly. The cells are correctly balanced and the voltage is good. The only problem I [...]

Same problem, the high-power mode will automatically shut down after ten seconds, a period of time later, it can resume by using the low gear mode.

tinfever, 2023-03-23:
What LED code are you getting when the vacuum shuts down after 10 seconds in high power mode?

slimlesaurien, 2023-03-23:
hello when the problem occurs the battery flashes blue 3 times and after a few minutes it works again then the problem comes back.
Cordially

tinfever, 2023-03-23:
Three blue flashes means:

3x Blue flashes - Battery low (Low voltage cutoff reached).
Output disabled until charger connected or pack goes to sleep and forgets.

It means your battery cells are reaching the 3V low voltage cutoff. Your battery cells probably have less capacity than they did when they were new. They also probably have a higher ESR than new. The high power mode draws much more current, which means the higher ESR of the cells causes the measured level of the cells to be lower under load.

If you wanted to, you could try lowering the MIN_DISCHARGE_CELL_VOLTAGE_mV parameter in config.h from 3000 to 2700 and you might get a little bit longer out of the cells, but there isn't much that can be done otherwise.

slimlesaurien, 2023-03-23:
hello sir can you explain to me how to modify this value on the.hex file, I am a bit lost I am a beginner and these are complicated for me.
Cordially

tinfever, 2023-03-24:
You'd need to download MPLABX and edit the source code and recompile to generate a new hex. I'm afraid you'll have to figure out the details yourself.

slimlesaurien, 2023-03-24:
hello sir I can not use the software, it is complicated for me.
could you send me the hex file with the modifications.
Cordially

slimlesaurien, 2023-03-24:
Bonjour monsieur j'ai des erreur avec le xc8 
[image]

tinfever, 2023-03-24:
I'm not sure those are actual errors since it says "info". Look at the text in the "Build, Load" tab in your picture.

Anyways, I recompiled with the 2700mV low voltage cutoff. The hex file is attached. Although I think you are very close to being able to do it all yourself. Change the .txt extension to .hex to flash.

[FU-DYSON-BMS_v1_Reduced_low_voltage_cutoff_2700mV.txt](https://github.com/tinfever/FU-Dyson-BMS/files/11066702/FU-DYSON-BMS_v1_Reduced_low_voltage_cutoff_2700mV.txt)

slimlesaurien, 2023-03-25:
bonjour monsieur je suis vraiment nul je n'arrive même pas a renommée un fichier txt en hex pouvez vous m'aider.
cordialement

tinfever, 2023-03-25:
Then either the battery cell capacity is too low or the ESR of the cells is too high. This is a battery cell issue and they probably must be replaced if you want to run at high-power mode.

You could do more testing if you wanted. Measure the voltage on each cell _simultaneously_ while you run the vacuum in high-power mode and see how low the voltage on the cells goes. See if one cell is going lower than the others under load. You can also measure cell ESR by applying a known current load and then measuring the voltage drop of the cell as the load is applied.

I can't provide further help I'm afraid, unless there is some indication there is a bug in my software.

slimlesaurien, 2023-03-27:
hello sir thank you for your help, the battery is well distributed but now when I vacuum after a while it flashes red 7 times.

The second thing when the battery is empty ( flashes 3 times blue), I but changed it, the battery flashes red 12 times red can you still help me.
Cordially

tinfever, 2023-03-27:
7 red flashes means the cell temperature has reached 60C and they are too hot. This is probably because the ESR of the cells is causing them to heat up more even when you use the vacuum on normal power mode. 

12 red flashes means the battery pack is too hot to charge. You may need to let it cool down before connecting it with the charger. The temperature limits allow the battery to go up to 60C when discharging but it won't charge unless the battery is less than 50C, so if the battery is at 55C it might have been discharging fine but then produce an error when you try to charge it.

NemoLee, 2023-09-08:
In addition, when I added the capacitor as mentioned in the eevblog video, the red light started blinking 16 times, indicating a restart of the power chip. Strangely, everything started working fine again when the green light came back on. However, after using it at high power and recharging, the issue of the blue light blinking 3 times occurred again.