tinfever/FW-Dyson-BMS #15: Dyson V7-SV11-PCB228499...maybe...dead

YorkYYY, 2022-11-05:
Hi,

    Thanks for made this project. It feels like give me a chance to fight against the sir James Dyson:) However, my bms maybe give t's life during the fight...
    As I'm a rookie... First I write your hex file with software pickit 3 programmer V1.0 which the seller provided, instead download from the link you provided. After wrote, the led on the BMS doesn't blink at all...
    Then i download pickit 3 programmer V3.10 from the link you provided. Still after i put a magnet, and release  the trigger, nothing shows on the led but the programmer can communicate to the BMS as your video shows. Then i try to write your hex file again, It shows successfully wrote, but the led still doesn't blink. I retried several times, Include erase first, then write, Still nothing on the led shows. and no voltage output when i release the trigger and put the magnet on.
    By the way, After the bms seems dead, I found the programmer can communicate to the BMS, even I hold the trigger and Take away the magnet, When i do so, And hit check communication, the led on the BMS blink a second.
    So that's my report, General... forgive my careless. And Is there anything I can do for this BMS, or just let it rest in peace......

tinfever, 2022-11-06:
Hi,

What was the original issue with your battery that prompted you to install this firmware? Knowing that might narrow down the issue.

It sounds like once the programmer is disconnected from your battery pack, there is nothing you can do that makes the LED light up, is that right? Including leaving a magnet over the magnet sensor and pulling and releasing the trigger?

What happens if you leave the programmer connected and the magnet in place and then pull the trigger?

I'm wondering if your ISL94208 IC isn't producing 3.3V like it should. The firmware currently always has an LED of some kind turned on whenever it is running, so if the LED isn't lighting up it means the PIC isn't running, but the PIC is clearly functional if you are able to program it.

You could try using a multimeter to measure between VCC and GND on the BMS without the programmer connected. I'd place a magnet on the sensor, put your multimeter probes on those points of the PCB, and then pull the trigger. You should ideally see 3.3V if the ISL94208 is regulating that properly.

https://github.com/tinfever/FU-Dyson-BMS/blob/main/hardware-info/images/PICkit%20Wiring%20Diagram_crop.jpg

YorkYYY, 2022-11-07:
Hi,
      Great thanks for your quickly reply, since I'm in a bad mod for losing this "fight" :)
      I want to apply this firmware because the cells go moderately 300mV out of balance, I think, when the red led flashed, I disassamble the battery pack and test the cells with multimeter and it shows 1 cell out of 6 got 3.6v while others are above 4.0V. I charged each to 4.2V separately and then try to install this firmware.
      Then things go like what I said...
      And today I using a multimeter to measure between VCC and GND on the BMS without the programmer connected, with magnet on or not, either pull the trigger or not, it's only 0V on the multimeter...
      And the programmer also CAN NOT communicate to the PIC today...while it still CAN YESTERDAY... which I'm sure of because I try to read, erase, and write the PIC several times after first failure...
      Anything I can do now...?

tinfever, 2022-11-09:
Yeah, it's sounding like the 3.3V onboard might not be working. I'm curious, if you leave your external DC power station providing 3.3V connected after writing the firmware, but with the programmer disconnected, does the LED on the BMS do anything when you press or release the trigger?

It sounds like this is exactly what you did but here is a photo of a V6 BMS and me measuring the 3.3V rail. I have the plastic trigger plunger partially unseated so it is like I'm pulling the trigger or continuously pushing that plunger down. On your V7 you'd have to also have a magnet on the reed switch (magnet sensor) for this to work but it sounds like this is what you did.
[image]

Barring any new information, I'd say the last thing you could try is just writing the firmware (by the way, in the PICKIT software there is a checkbox to have the programmer supply 3.3V so you don't have to use the external DC power station when using the programmer), putting the battery back in the vacuum, and seeing if anything changes. It's unlikely but the magnet placement can be a little tricky so it's possible not having the magnet perfectly placed would give you similar issues to what you are seeing I think. Putting the battery back in the vacuum, even without installing the screws, would quickly eliminated that (remote) possibility.

If none of that works, I think probably the 3.3V regulation on the ISL94208 is broken. I've managed to break that part of the chip on multiple battery packs of mine during development, so I think it may be particularly fragile. If that is indeed what has failed, unfortunately the only remedy is probably replacing the ISL94208 IC. I'm still not sure quite why this part fails this way, although my best theories (not based on much) is that maybe it could be ESD or voltage spikes on the 3.3V rail when connecting a programmer, or maybe when the programmer is configured to also supply 3.3V it is hard on the ISL94208. Like I said, I've broken several of them so it's easy to do somehow and I'm not quite sure how it happens.