tinfever/FW-Dyson-BMS #19: Dyson V8 -  BMS  180207-01/04 - Battery SV10

accampato, 2023-01-02:
Hi all,
this message only to say thank you to the amazing work of **tinfever**.
I have tried to delete only EEPROM after cells rebalancing, but I deleted the entire PIC, so i put tinfever firmware and all works perfect again.
My dead dyson battery in now alive again. The firmware works perfect also on Dyson V8, the only change is on the LEDs code.

My very small contribution to this project is the conversion of LEDs color code from V6 to V8.
Assuming the three LEDs in V8 version numbered 1, 2 and 3 according to the following image:

[image]

Here is the conversion matrix:
[image]
So refer to https://github.com/tinfever/FU-Dyson-BMS#what-do-the-leds-mean to see **what do the LEDs mean**

Great job, thanks so much again to FU planned obsolescence.

nepletkai, 2023-03-02:
Hi, yesterday successfully flashed the Dyson V8 SV10 battery (exactly the same part number) as per the example above (which is used on Dyson V7 vacuum). The only question is - does there is a plan to adopt the code for proper 3 LED operation on SV10 batteries?
p.s. SV10 battery case with small plastic body trimming fits Dyson V7 vacuum ( if this was the only issue that stops for testing and development so far)
[image]

tinfever, 2023-03-03:
> Hi, yesterday successfully flashed the Dyson V8 SV10 battery (exactly the same part number) as per the example above (which is used on Dyson V7 vacuum). The only question is - does there is a plan [...]

Thanks for the tip on fitting a V8 battery in a V7 vacuum, and the pictures. One day I want to add support for the correct LEDs on the V8, among other things, but I have no idea when I'll get around to it. It could be years. :(

iwaitlyd, 2023-03-30:
> Hi all, this message only to say thank you to the amazing work of **tinfever**. I have tried to delete only EEPROM after cells rebalancing, but I deleted the entire PIC, so i put tinfever firmware [...]

Thanks to tinfever for the great job to make dyson battery alive.

And Thanks to accampato for decode the LED color on V8.

After reading your post, I think just replace the 3rd Blue LED to green will fix the color code, as the picture. Extend it a little bit so it will illuminate form the 1st window. Red+Green=Yellow, Red+Green+Blue=White.
[image]
[image]

jayshin, 2023-07-26:
> Thanks for the tip on fitting a V8 battery in a V7 vacuum, and the pictures. One day I want to add support for the correct LEDs on the V8, among other things, but I have no idea when I'll get around [...]

> > Thanks to tinfever for the great job to make dyson battery alive.
> > And Thanks to accampato for decode the LED color on V8.
> > After reading your post, I think just replace the 3rd Blue LED to green will fix the color code, as the picture. Extend it a little bit so it will illuminate form the 1st window. Red+Green=Yellow, Red+Green+Blue=White.
> 
> UPDATE: I replace the LED successfully. The green color shows up correctly. Since my battery cell are balanced, I have not seen yellow yet.

Hi iwaitlyd, could you share your code changes/or the firmware you built for V8? I'd like to test it on my V8 batter.
Thanks!

LexX71, 2024-10-27:
help me solve the issue. when connecting a native power supply. the battery flashes 8 times. when connecting the laboratory 1a power supply, everything is fine