tinfever/FW-Dyson-BMS #21: V6, don't know how to identify the version of the PCB，and don't know whether the battery is locked or not

kanelau, 2023-01-13:
The Dyson doesn't work when holding the trigger. The red light flashes. But when I detach the battery from the vacuum cleaner, then press the button on the battery, the battery shows blue, long press it turns green. Is my battery locked or not? 
[image]
[image]
The battery with the lowest voltage is 3.9v, and the battery with the highest voltage is 4.1v.

I can't find the version number from battery and its PCB. But find a QR code on the PCB, the scanning result is 188055010416C096931VX196-17021602--U

If you have time, appreciate your response.

Tam

Smtblackstone, 2023-01-13:
Şeffaf parçanın hemen altında sağ taraftaki konnektör tarafında

kanelau, 2023-01-14:
> Şeffaf parçanın hemen altında sağ taraftaki konnektör tarafında 

Many thanks. Do you think my battery is locked by PCB or not? Would it help to burn the author's firmware?

Smtblackstone, 2023-01-14:
Hücrelerin hepsini tek tek şarj etmelisin 4.1 veya 4.2 v olacak şekilde daha sonra paylaşılan yazılım dosyasını yükleyerek denemelisin ben 1 den fazla V6 ve v7 için denedim çalıştı sorunsuz bu işlemlerden sonra deşarj ederek yük yesti yap ve pillerin voltajlarını tekrar ölç eğer hücreler arası dengesizlik varsa hücre değişimi yapmalısın

Smtblackstone, 2023-01-14:
Muhtemelen dengesiz voltaj kaynaklı PCB kitlenmistir zaten bildiğim kadariyla PCB yazılımı buna bağlı yazılmış sanırım

kanelau, 2023-01-14:
Many thanks. I will first use the pickit to burn a new firmware, and then test it again.

Smtblackstone, 2023-01-14:
Rica ederim bol şans

kanelau, 2023-01-17:
called Dyson service, they told me 10 times red light flashing means the motor had got a problem. So i bought a 2nd hand motor head. Unfortunately it doesn't work too, the battery still flashes red light 10 times when press the trigger. But if detach the battery from the mainbody, the battery light is blue and green. Someone can give your expertise support? Many thanks.
[image]

kanelau, 2023-01-17:
I have flashed tinfever's firmware. It doesn't work too. It turns to 16 red blinks if attach battery to the main body and press the trigger. But if i detach the battery from the main body, the light is red-blue-green. Obviously i am in the error "ISL_BROWN_OUT", i don't how to get it resolved. Appreciate your response and comments.

Smtblackstone, 2023-01-17:
Görselde b+ nikel bağlantısında hasar mevcut sizmi yaptiniz yoksa kendimi oluştu

Smtblackstone, 2023-01-17:
Batarya bloğunu harici yük testine tutma şansınız varmı yük testinde yaklaşın 15 amper yüklenerek bataryanın güç durumuna bakabilirseniz sonuca daha yaklasabiliriz

kanelau, 2023-01-18:
Yes, I got a short circuit while trying to power the vacuum cleaner directly with the wires. I still don't understand why the short circuit occurs.  In fact, as soon as the battery is connected to the host, the error signal of the battery is a short circuit, which was the case at the beginning. The short circuit didn't damage the battery nor the PCB.

Smtblackstone, 2023-01-18:
Yani bu durumda süpürgenin motoru veya motor arkasinda bulunan elektronik kart arizali sanırım

kanelau, 2023-01-18:
> Yani bu durumda süpürgenin motoru veya motor arkasinda bulunan elektronik kart arizali sanırım 

Many thanks for your advice. But I have bought another 2nd hand motor head, which had been tested ok before the seller dilivered it to me. The only possibility i haven't test is the problem caused by imbalance. I do have balance charger.

dabaoge666, 2023-05-12:
Hello, I have a Dyson V10 battery flashing red light, I want to re-flash the firmware, where should I find the firmware