tinfever/FW-Dyson-BMS #25: Programming with PICKIT2 successful

ibarrajo, 2023-03-07:
Just as a note for people who have a PICKIT2 and not the PICKIT3.

PICKIT2 does not support PIC16LF1847 and will refuse to communicate with it.

Nonetheless, I was able to flash this firmware successfully by updating my `PK2DeviceFile.dat` in `C:\Program Files (x86)\Microchip\PICkit 2 v2` with the device file provided in the software for PICKIT3.

`PIC16LF1847` is detected and works

[image]

liujixu1984, 2023-12-06:
Hello, I have a Pickit2 and Dyson V6 battery. Can you please send me an explanation of your successful use of Pickit2 to flash the device? thank you.

ibarrajo, 2024-03-14:
I'm going to close the issue, since I just intended to communicate my findings. 

To be clear, I just updated my `PK2DeviceFile.dat` used by PICkit2 by replacing it with the same file that comes with Microchip's PICkit3 Programmer Application. I did this because I didn't want to buy a PICkit3 when I just had a PICkit2 already.

closed 2024-03-14 by ibarrajo