tinfever/FW-Dyson-BMS #26: When to flash the firmware?

Tharrington86, 2023-03-09:
I am only getting about 1 minute of run time out of my v7's battery pack so I am going to replace the cells. When would be the optimum time to flash the new firmware? Before replacing the cells or after?

tinfever, 2023-03-09:
I'd flash the firmware afterwards, especially since you might not end up needing to flash it. It's possible you could use the EEPROM edit mentioned here to unlock your pack and keep the factory firmware: https://www.eevblog.com/forum/reviews/dyson-v7-trigger-cordless-vacuum-teardown-of-battery-pack/msg4028665/#msg4028665

Tharrington86, 2023-03-10:
@tinfever thank you! I had not seen the unlock yet. I will give that a shot first.