tinfever/FW-Dyson-BMS #30: Works on Dyson V8 (YH5)

justinmassiot, 2023-06-20:
I'm reporting that your firmware works on my Dyson V8, battery model YH5. I programmed it through **PICkit4 and MPLAD IPE 6.10**.
Unfourtunately, I erased the stock firmware of my BMS, so yours was my last chance 😄 Now I missed a correct behaviour for the LEDs...

I can give you the exact PCB number and revision if useful. I can even take pictures to document the V8 battery.

Thank you very much for the awesome work!

edreyyo, 2023-09-19:
@justinmassiot Is your battery still working well? I'm thinking about flashing this firmware to my Dyson V8.

justinmassiot, 2023-09-24:
@edreyyo unfortunately no. But I can't tell for sure if it's a battery problem or a firmware issue.

If you have a "32 red flashing LED" error, I suggest you try the "error reset" trick first: https://www.youtube.com/watch?v=7t9yA3_KUI0 **but** it may not be applicable to the V8.

edreyyo, 2023-10-06:
Thanks for the reply @justinmassiot. I'll try the "error reset" trick

turenabo, 2024-02-21:
> @edreyyo unfortunately no. But I can't tell for sure if it's a battery problem or a firmware issue.
> 
> If you have a "32 red flashing LED" error, I suggest you try the "error reset" trick first: https://www.youtube.com/watch?v=7t9yA3_KUI0 **but** it may not be applicable to the V8.

Tinfever has another project https://github.com/tinfever/open-dyson-battery , there is a V8 schematic, I compared with v6 18002, there are some different.  eg, one more gpio is used for LED, shunt is 0.001 but not 0.002... I think code should be modified to adapt to v8. Do you know where we can see MCC of this project? MCU should be configed accordingly, but I don't know how to find mcc.

Nickduino, 2024-05-27:
> Tinfever has another project https://github.com/tinfever/open-dyson-battery , there is a V8 schematic, I compared with v6 18002, there are some different. eg, one more gpio is used for LED, shunt is [...]

The original code allows you to choose the version:
https://github.com/dr-mark-roberts/open-dyson-battery/issues/4#issuecomment-1120256189

HarmOtten, 2024-08-26:
I also flashed a Dyson V8 battery YH5 BMS to FU-Dyson-BMS. Directly after programming it seems to work. But when the battery falls asleep, it will not awaken when pressing the trigger or connecting the charger...
The BMS version is: 180207-02/01

@justinmassiot Maybe you should change the title of this issue to: "Does NOT work on Dyson V8 (YH5)"

@tinfever Please, PLEASE, tell us how we can convince you of making a separate version of this firmware for the V8 !