tinfever/FW-Dyson-BMS #31: strange red flashes

jnpa007, 2023-06-27:
Great job thanks again for this hack.
I have a problem with several v6 sv03 battery pack. After replacing the cells with sony vtc6 and flashing the pcb to successfully remove the red flash error, when I used the pack until exhaustion it was impossible for me to recharge it with the European mains charger but that works with the 12 volt car charger. It's strange, both deliver a very close voltage, only the maximum current is different.

[image]
[image]
[image]
[image]

see video here:
https://www.youtube.com/watch?v=Vo0QCxLlwVQ

Please help :)

jnpa007, 2023-06-27:
The 5ms Firmware seems to fix the problem!! 
Done.

kulic23, 2024-04-26:
Hi, how do I solve the problem with 8 red flashes while charging?

> Прошивка 5 мс, кажется, решает проблему!!Готово.