tinfever/FW-Dyson-BMS #33: DYSON V7 SV11 228499 0103

chaddyyyy, 2023-08-17:
The battery works fine until i connect to the Dyson,, i get 16 reds blinks (brownout) when pulling the trigger,
when i disconnect the battery from the dyson, the battery is fine again lights act as they should,
no problems with the motor in the dyson as i have other batteries(original firmware) that are okay with it,
is there anything i can alter in the firmware?
thanks

{
    "Firmware": "Tinfever FU-Dyson-BMS V1",
    "Total_Runtime_Seconds": 5.408,
    "Faults": [
        {
            "index": 0,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 5.408
        },
        {
            "index": 1,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 5.408
        }

tinfever, 2023-08-19:
If the battery pack operates fine outside the vacuum, but gives the ISL_BROWN_OUT error when inserted, my current best guess is that the cells have too high of an internal resistance (due to age or wear) such that the inrush current of trying to start the motor causes the cell voltage to drop enough that the ISL94208 chip turns off.

I think I've written some more about it in the EEVBlog forum thread, but there is no good solution since it is a battery cell issue. I have a theory that maybe if you soldered in an extra capacitor to specific pins on the ISL94208, it might be able to ride through the inrush current caused voltage drop, but this is untested, and would be a hardware mod.

chaddyyyy, 2023-08-21:
@tinfever sorry for the late reply, didn't wanna reply with no info :)
I've scanned over the eevblog, and read the conversation between you and Bluesight...

i did what you said 10uf 16v cap from the blue circled R48 to the ground of D3 
and can confirm it is working!! just need to tidy it up but heres a pic of the mod,
[image]

i always knew the batteries were pants, I just didn't understand how dramatically someone could put an IC together or should i say how Dyson had configured it... id rather have an underpowered motor than a stupid fault reset,
anyways thanks for the push in the right direction :)

tinfever, 2023-08-23:
That's great! Thanks for trying it and reporting back. I'm happy to hear it worked!

djodjo02130, 2025-01-06:
Hi @tinfever @chaddyyyy,

I have exactly the same board 228499.

Thanks a lot i fix my issue with an 25v 10uf ( didn't have 16v 10uf ).

[image]
[image]