tinfever/FW-Dyson-BMS #34: Original Firmware Fix

chaddyyyy, 2023-09-15:
if this is not allowed tinfever feel free to remove this,

some useful info here lads...
31 flashing red lights,

[image]
hook up the BMS.. no need to wake the battery up, if anything let it sleep.. or leave it to complete its red light blinking if triggered, let it fall alseep, then connect,

[image]
make sure program memory is unchecked "disabled" or you will brick the original firmware,

[image]
click auto import hex + write

[image]
and find the working HEX you need,

downloads here.. extract the zip 

[DC59 61462-0107.zip](https://github.com/tinfever/FU-Dyson-BMS/files/12614265/DC59.61462-0107.zip)
[SV03 188002 0104.zip](https://github.com/tinfever/FU-Dyson-BMS/files/12614266/SV03.188002.0104.zip)
[SV04 188002 0104.zip](https://github.com/tinfever/FU-Dyson-BMS/files/12614267/SV04.188002.0104.zip)
[SV10 180207-0104.zip](https://github.com/tinfever/FU-Dyson-BMS/files/12614268/SV10.180207-0104.zip)
[SV11 228499 0103.zip](https://github.com/tinfever/FU-Dyson-BMS/files/12614269/SV11.228499.0103.zip)

notes..
if you still get 31 red flashes there's a problem with your batteries and needs more investigating, ie swopping for better ones,

if it works but when you connect to the dyson you get 10 red flashes when pulling the trigger, The voltages on the battery cells are dipping too much in voltage, you will need to connect a 10uf 16v cap to D3 & R48 like this..
[image]

i have revived loads of boards back to life.. not guarantee it will work for you, but from experience.. the boards I was not able to revive also had problems with aftermarket firmware with 16 red flashing.. brown outs, battery acid damage to the board etc etc

just quickly built this and shes a beast,
1701mah to 1750mah top row
1950mah to 2010mah bottom row, all holding at 41.50v to 41.80v after sitting for few weeks
still kept getting the 10 red flashing light on trigger so had to mod it.. i get V8 vibes from it good suction and long battery life :)
[image]

Smtblackstone, 2023-09-23:
Cihazın LED ışıklarında herhangi bir farklılık varmı acaba ? Teşekkürler

chaddyyyy, 2023-09-23:
> Is there any difference in the LED lights of the device? Thanks

no this is the original firmware

Smtblackstone, 2023-10-01:
SV 11 zip hatası veriyor . Arşiv hasarlı hatası .yardımcı olabilirmisin teşekkürler

chaddyyyy, 2023-10-01:
> SV 11 zip hatası veriyor . Arşiv hasarlı hatası .yardımcı olabilirmisin teşekkürler
@Smtblackstone 

rename the ZIP to RAR, GitHub won't allow RAR so I renamed it to ZIP, this might be the problem, also uploaded an HH11 dump for the V7 if the SV11 does not work.

Smtblackstone, 2023-10-01:
Teşekkürler @chaddyyyy

turenabo, 2024-02-03:
many thanks, have you tried on V8? hope you see this comment and bring us a V8 original firmware.

chaddyyyy, 2024-02-03:
@turenabo SV11 is for the V8 🙂

turenabo, 2024-02-04:
hi, maybe I misunderstood what you meant, I checked your file, they are all several hundred bytes, Tinfever's hex file is 40k bytes, so let me try to understand, your way is to add something into locked BMS' EEROM then unlocked the battery? thanks.

chaddyyyy, 2024-02-04:
@turenabo yes import hex to eeprom data, make sure program memory is not enabled, check the pictures 🙂

v47stepanenko, 2024-09-14:
worked flawlessly for my SV10 180207. appreciate it!

zzp8202, 2024-10-28:
Thanks for share, it's worked on SV10K, before reset, it flash red light 4 times.

erayteknik07, 2026-09-09:
v12 bms yazılım indiremiyorum