tinfever/FW-Dyson-BMS #35: Flashed - was working now not

ShukranPovo, 2023-09-27:
Thanks for the guide and work.

I flashed my SV11 pack from my V7 fluffy with your V1 firmware and it was working as expected for a couple of days. Now it only runs for 15 seconds and flashes 3 blue with no output when in the vacuum. On the bench with a magnet it performs as expected with 4 to 5 green flashes when released indicating charge is ok.

tinfever, 2023-09-27:
The three blue flashes are the low battery cell voltage indicator. One or more of the battery cells might be dropping too low under load. Odd that it was working fine for a few days though. You're not using the vacuum in Max Power mode, are you?

ShukranPovo, 2023-09-30:
It was on Max power when it was running for a few seconds. Now does not run at all. On the bench the battery flashes 5 green when I release the button and when I put it on charge the yellow only flashes 4 times. When in the machine in Max or normal power it flashes 16 red as soon as I pull the trigger. On the bench when I push the button I get 23.84V.

ShukranPovo, 2023-09-30:
Just opened the battery again - the cells are at 4.00, 3.95, 3.90, 3.95, 3.91, 4.10. and the output is at 23.84 at the terminals.

ShukranPovo, 2023-09-30:
I have run the EEPROM Parsing tool and got the following errors. Now what? :-)

    "Firmware": "Tinfever FU-Dyson-BMS V1",
    "Total_Runtime_Seconds": 5088.032,
    "Faults": [
        {
            "index": 0,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 2804.096
        },
        {
            "index": 1,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3228.864
        },
        {
            "index": 2,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3228.864
        },
        {
            "index": 3,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3628.16
        },
        {
            "index": 4,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3725.952
        },
        {
            "index": 5,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3725.952
        },
        {
            "index": 6,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3725.952
        },
        {
            "index": 7,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3725.952
        },
        {
            "index": 8,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3725.952
        },
        {
            "index": 9,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3725.952
        },
        {
            "index": 10,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3736.64
        },
        {
            "index": 11,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3938.368
        },
        {
            "index": 12,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3993.28
        },
        {
            "index": 13,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 3993.28
        },
        {
            "index": 14,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 4011.968
        },
        {
            "index": 15,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 4029.12
        },
        {
            "index": 16,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 4890.752
        },
        {
            "index": 17,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 4902.208
        },
        {
            "index": 18,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 4972.192
        },
        {
            "index": 19,
            "error_meaning": [
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "Trigger",
            "timestamp": 4997.248
        },
        {
            "index": 20,
            "error_meaning": [
                "THERMISTOR_OVERTEMP_PICREAD",
                "UNDERTEMP_FLAG",
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "None",
            "timestamp": 5088.032
        },
        {
            "index": 21,
            "error_meaning": [
                "UNDERTEMP_FLAG"
            ],
            "detect_mode": "None",
            "timestamp": 5088.032
        },
        {
            "index": 22,
            "error_meaning": [
                "THERMISTOR_OVERTEMP_PICREAD",
                "UNDERTEMP_FLAG",
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "None",
            "timestamp": 5088.032
        },
        {
            "index": 23,
            "error_meaning": [
                "THERMISTOR_OVERTEMP_PICREAD",
                "UNDERTEMP_FLAG",
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "None",
            "timestamp": 5088.032
        },
        {
            "index": 24,
            "error_meaning": [
                "THERMISTOR_OVERTEMP_PICREAD",
                "UNDERTEMP_FLAG",
                "ISL_BROWN_OUT"
            ],
            "detect_mode": "None",
            "timestamp": 5088.032
        }
    ]
}
Press <ENTER> to continue

ShukranPovo, 2023-09-30:
Now it runs for a while till I stop then it flashes red for a while and does not work until it has timed out, then it works till I stop again - on and on, battery is reassembled so I can not read the eeprom again.

tinfever, 2023-09-30:
Thanks for all the info. For future reference, when you say "flashes red for a while", knowing the exact flash codes would be more helpful.

It sounds like the cells are probably marginal and so the ISL94208 is browning out during the startup inrush current. If the cells have a higher ESR due to wear/age, they'll measure fine when not under load, but the internal resistance will cause the voltage to drop under load. If the ESR is high enough, the voltage under no load can be 4.2V but when you apply the 20A load of the vacuum, the cell voltage could immediately drop below 3V triggering the 3 blue flashes to say the cell voltage is too low. If the cell voltage drops even lower, it can cause the ISL94208 to brown out and reset, which triggers the ISL_BROWN_OUT error.

If the cells are only marginal, others have had success adding a capacitor in a specific location so the ISL chip can ride through the startup inrush current demand. More info in the links below, although I think even a 1uF ceramic cap would be fine instead of the larger caps others have used.

However, if the cells are bad enough, this cap won't fix the issue. Ideally, I'd recommend testing the cells under a known load and calculating their internal resistance.

https://www.eevblog.com/forum/projects/fu-dyson-bms-an-(unofficial)-firmware-upgrade-for-dyson-v6v7-vacuum-bms/msg4756034/#msg4756034

https://github.com/tinfever/FU-Dyson-BMS/issues/33

ShukranPovo, 2023-09-30:
Thanks for the reply - sorry, "flashes red for a while" means 16 red flashes.

It works erratically, I am assuming the cells have a marginally high internal resistance as you suggest (I have no idea how to calculate this or test for it) - the pack is at least 6 years old and used to be used every day. The one issue I have is that I am in South Africa and nobody will ship a new pack to me - maybe I need to change out the cells for new ones but I have no spot welder and am weary of messing with lithium cells - I am a noob to this, I used to replace NiCad cells but have not messed with Li before.

tinfever, 2023-09-30:
I would try the capacitor mod if you can. That would be attaching a 6.3V 10uF or better capacitor between the two points marked in blue (the indicated terminal of R48 connects to the VBACK pin on the ISL94208, and the indicated pin on D3 connects to ground). Lesser capacitance might work as well, given the vacuum mostly works under normal conditions with no cap at all.

[image]

If you wanted to measure the cell resistance, you'd need to apply a controlled load of something like 1A to a single cell in the pack, and then measure the difference in voltage with and without the load applied. Don't apply the load for too long since that will drain the cell which would also cause the measured voltage to drop. You could apply the load with a high enough rated resistor, like 5.1R rated for at least 5-10W in series with one multimeter in current measurement mode. Then have another multimeter measuring the voltage across the cell.

If when you apply the load (we're talking for a few seconds here) the measured voltage is 3.5V and the current is 900mA, and when you remove the load the voltage instantly (don't wait since cell voltages will naturally rise slowly after load removal too) goes to 3.6V: (3.6V-3.5V)/0.9A = 111 milliohms cell resistance.