tinfever/FW-Dyson-BMS #36: Bizarre SV03 issue

dickeos, 2023-09-27:
Please can someone help me understand the issue I have come across, its making me go crazy!

I have replaced all the cells in an ageing SV03 battery pack (with Sony equivalents), flashed the pack with the tinfever firmware, done a power cycle and all appeared great, like it has done several times before with other packs I have refurbished.

I then put the pack back together and connected it to the V6, pulled the trigger and got 16 red lights. I then opened the pack back up and used jumper leads to hook it up the the V6 from the battery terminals and it works perfectly. I have reduced the length of the jumper leads down to as small as I can and it still works, but when I plug the battery directly into the V6 I get the red lights.

Now I know 16 means a brown out so I checked and double checked for a short on the terminals and there isnt.

I have no idea what it could be, its feels like its proximity thing but common sense tells me it cant be.

tinfever, 2023-09-27:
The inductance of the jumper leads you are adding might be slowing down the vacuum inrush current enough to avoid the brown out of the ISL chip. Although, usually the brown out issue only occurs when the cells have a high ESR so the voltage dips too much, or if the load is a direct short circuit.

In the EEVBlog forum thread, and I think in another github issue here, I've discussed where you can try adding a capacitor on the control board to ride through the voltage drop during startup, and someone has tested it and found that to work. I should really only be applicable to high ESR (usually worn) cells though. You might use a scope to watch the voltage on those points and see if you are getting a large supply voltage dip going to the ISL chip during startup, and then maybe work from there. I'd double check your cells though.

dickeos, 2023-09-28:
Thank you for the hints and advice. Ill try the capacitor trick and if it fails I do have a battery tester which I can use to test the cells. Ill report back with the results.

dickeos, 2023-09-30:
That recommendation to use a capacitor worked. Strange thing is that the battery's all tested ok, that said i want discharging them at the rate the dyson is.

tinfever, 2023-09-30:
I'm glad to hear it worked! How are you testing the cells?

dickeos, 2023-10-22:
Hi I’m using an internal resistance tester and the batteries are all around the 50 m ohm range.

interestingly I bought some 25a LG cells as an alternative to the Sony ones and I had the same results on a SV03 but on a dyson v7 battery they worked.

tinfever, 2023-10-22:
Interesting. I can't explain that. Maybe the SV03 vacuum motors have a higher inrush current? Maybe 50 mOhm is still too high? I'm not sure. The original batteries are specified for less than 20 mOhm AC impedance but I'm not sure how that translates to what we're able to measure, or how high it can go in most stock vacuums before issues occur.

https://github.com/tinfever/FU-Dyson-BMS/blob/main/hardware-info/Li-Ion%20Cell%20Spec%20Links.txt