tinfever/FW-Dyson-BMS #38: v6 188002

liujixu1984, 2023-12-09:
Hello, I have a v6 battery with PCB number 188002. The battery has been manually balanced and the v1 version hex file has been flashed. Now, pulling the trigger will flash the blue light three times, and inserting a charger will flash the yellow light for about 50 times. However, the voltage of each battery has been measured and is normal. I skipped the protection board battery and directly connected to the main unit, which works. I have discharged some of the electricity and am now plugging in the charger, which shows a blue light status. Can this make it work normally?

tinfever, 2023-12-09:
Usually if you are getting the 3x blue flash code when pulling the trigger even after fully charging the battery, it could be because one or more of the cells have too high of an internal resistance, and so the measured cell voltage is dropping too much under the load of the vacuum. One way to test this is to try enabling the battery with no load connected (not in the vacuum). If the output turns on fine with no load, but gives the low battery voltage code with a load, that would indicate a battery cell issue. This would also explain why it works when you bypass the BMS board, since that would be bypassing the under voltage cutoff. You could also try measuring the voltage of each cell while you have the vacuum connected bypassing the BMS board. Then you can see how low the cell voltage is under load.

However, I'm concerned about the 50 yellow light flashes when connecting the charger. Could you elaborate more on this? Does it happen only when you plug in the charger and not when it is removed? What happens after the yellow flashes? 50 yellow flashes would normally indicate a 2500mV cell imbalance, which is extremely bad. However, you mentioned you'd checked the voltage of each battery cell and they were fine. So if the cells are actually at a fine voltage, but the BMS is reporting such a significant imbalance, that could indicate a failure of some part on the BMS board. Perhaps the BMS isn't actually reading one of the cell voltages properly because of a bad solder joint, cracked resistor, or something.

liujixu1984, 2023-12-10:
> 通常，如果您在电池充满电后扣动扳机时仍收到 3 倍蓝色闪烁代码，这可能是因为一个或多个电池的内阻过高，因此测得的电池电压在真空负载下下降太多。测试这一点的一种方法是尝试在未连接负载的情况下启用电池（而不是在真空中）。如果输出在空载时正常开启，但在负载时给出低电池电压代码，则表明电池单元有问题。这也可以解释为什么当您绕过 BMS 板时它会起作用，因为这将绕过欠压截止。您也可以尝试在绕过 BMS 板连接真空时测量每个电池的电压。然后你可以看到电池电压在负载下有多低。
> 
> 但是，我担心连接充电器时黄灯闪烁 50 次。你能详细说明一下吗？它是否仅在插入充电器时发生，而不是在拔下充电器时发生？黄色闪烁后会发生什么？50 次黄色闪烁通常表示 2500mV 电池不平衡，这是非常糟糕的。但是，您提到您已经检查了每个电池单元的电压，它们都很好。因此，如果电池实际上处于良好的电压状态，但BMS报告了如此严重的不平衡，则可能表明BMS板上的某些部件出现故障。也许 BMS 实际上没有正确读取其中一个电池电压，因为焊点坏、电阻器破裂或其他原因。

> Usually if you are getting the 3x blue flash code when pulling the trigger even after fully charging the battery, it could be because one or more of the cells have too high of an internal [...]

Yes, when inserting the charger, the yellow light will flash 59 times, then display the blue light and start charging. After removing the charger, the yellow light will also flash 59 times. The previous test was conducted inside the vacuum cleaner and with the battery removed, and in both cases, the blue light will flash 3 times. When measuring the voltage of each battery cell with a multimeter in the unloaded state and bypassing the protection board to power the host, All are 4.1xx under no-load conditions and 2.7xx under load vacuum cleaner max mode, so there is no particularly large voltage deviation of any battery cell. Does this indicate that the battery protection board is faulty?

liujixu1984, 2023-12-10:
After my inspection, it turned out that the R24 resistor was knocked off when disassembling the shell. After supplementing the original components, the battery is now working properly. Thank you very much!

tinfever, 2023-12-11:
That's excellent news! I'm glad you got it figured out. Thank you for reporting back.

closed 2023-12-11 by tinfever