tinfever/FW-Dyson-BMS #39: v6 Low temperature protection

liujixu1984, 2023-12-18:
At around 15 degrees Celsius at room temperature, the battery will flash 14 times to display low temperature protection. May I ask what is the low temperature protection temperature for your battery? Is there any way to make him work? Except for heating up

tinfever, 2023-12-18:
I was actually just answering a YouTube comment about this the other day. It's possible the original firmware didn't have low temperature protection or maybe their threshold was set lower. The threshold should be roughly 7C. If you wanted to, you could remove that protection by commenting out the two lines of code in the safetyChecks function that reference MIN_TEMP_C, I think.

I'm not sure why that error is occurring for you at 15C though. It could be that I got the thermistor beta value wrong.

liujixu1984, 2023-12-21:
> I was actually just answering a YouTube comment about this the other day. It's possible the original firmware didn't have low temperature protection or maybe their threshold was set lower. The [...]

Because I don't know how to compile, could you please modify and cancel the low-temperature protection before sending a hex file?

thinker226, 2024-01-24:
> Because I don't know how to compile, could you please modify and cancel the low-temperature protection before sending a hex file?

I faced the same problem and couldn't solve it by myself, looking forward to the author's update.

turenabo, 2024-01-25:
same problem, it doesn't work at 11c, it recovered at 15c，Ithink 11 is not too low, 7 is reasonable, or maybe no protection is more better.

turenabo, 2024-02-03:
> I was actually just answering a YouTube comment about this the other day. It's possible the original firmware didn't have low temperature protection or maybe their threshold was set lower. The [...]

Hi, Tinfever, do you know what does orignal firmware 14 red light flashing mean? one of my 61462 get 14 red flashing and the board was locked. I am interested in why does it lock? thanks

tinfever, 2024-02-03:
> Hi, Tinfever, do you know what does orignal firmware 14 red light flashing mean? one of my 61462 get 14 red flashing and the board was locked. I am interested in why does it lock? thanks

I don't know the reason for the 14 red flash error code on the original firmware.