tinfever/FW-Dyson-BMS #40: How to compile firmware

liujixu1984, 2023-12-29:
May I ask how to compile firmware? I will use Git to clone it locally in Ubuntu environment, then enter the project folder firmware, and then run it/ Makefile command prompt:
root@ubuntu:~/FU-Dyson-BMS/firmware# ./Makefile                                                             

./Makefile: line 54: build:: command not found                                                              

./Makefile: line 56: .build-pre:: command not found                                                         
./Makefile: line 59: .build-post:: command not found                                                        
./Makefile: line 64: clean:: command not found                                                              
./Makefile: line 66: .clean-pre:: command not found                                                         
./Makefile: line 74: .clean-post:: command not found                                                        
./Makefile: line 79: clobber:: command not found                                                            
./Makefile: line 81: .clobber-pre:: command not found                                                       
./Makefile: line 84: .clobber-post:: command not found                                                      
./Makefile: line 89: all:: command not found                                                                
./Makefile: line 91: .all-pre:: command not found                                                           
./Makefile: line 94: .all-post:: command not found                                                          
./Makefile: line 99: help:: command not found                                                               
./Makefile: line 101: .help-pre:: command not found                                                         
./Makefile: line 104: .help-post:: command not found                                                        
./Makefile: line 110: include: command not found                                                            
./Makefile: line 113: include: command not found 

May I ask if there is a help file that teaches me how to do it?

tinfever, 2023-12-29:
I think you'll have to use MPLABX IDE from Microchip. I think there is a Linux version. Once you have that installed, it should be straightforward to open the project and then run/compile.

liujixu1984, 2023-12-30:
Thank you very much! I have already used the firmware with the comment of the lowest temperature!

tinfever, 2023-12-30:
That's odd. Were you using the [Initial Release - Version 1](https://github.com/tinfever/FU-Dyson-BMS/releases/tag/release-v1) hex file previously? Could you test again to compare both the [Initial Release - Version 1](https://github.com/tinfever/FU-Dyson-BMS/releases/tag/release-v1) release and the one you've compiled with no changes, and confirm they behave differently? They should be the same.

liujixu1984, 2023-12-31:
Yes, I'm attaching the compiled file without any changes, please compare it

[firmware.production.hex.txt](https://github.com/tinfever/FU-Dyson-BMS/files/13799939/firmware.production.hex.txt)