tinfever/FW-Dyson-BMS #41: Why temperature sampling circuits are different on 188002 and 61462?

turenabo, 2024-02-03:
Hi Thinfever, many thanks for your project, I am learning BMS through your project, it really help me a lot.

I compared 188002 and 61442 temperature sampling circuit, they are different, one is NTC+ operational amplifier, another is just NTC, the divider resister on first one is 6.2k, another one is 24k. 

So, CPU will get different voltage? can your firmware work on these 2 boards correctly?

many thanks again.

turenabo, 2024-02-03:
I think I found the answer in main.c, let me learn your codes carefully, many thanks

tinfever, 2024-02-03:
Hi, My firmware is able to detect which vacuum model is used so it can work seamlessly with all of the supported models. It uses different thermistor lookup tables depending on which model of vacuum it detects. I don't know why the original designers of the BMS board changed the design. I believe the opamp is being used as a comparator to send an overtemperature signal to the ISL94208 which sort of overrides the ISL94208's built-in overtemperature threshold. Perhaps the the designers wanted a different overtemperature threshold for the ISL94208's protection.

turenabo, 2024-02-04:
> Hi, My firmware is able to detect which vacuum model is used so it can work seamlessly with all of the supported models. It uses different thermistor lookup tables depending on which model of vacuum [...]

hi, thanks for your answer, yes, I saw it in main.c. Btw, I remember you commented maybe orignal firmware doesn't have low temp protection, I did an experiment today, I put a battery pack with orignal firmware in my fridge, it was -18c, it flashed 6 yellow  light when I push the button, no output, and it recovered when temp came up.

So, maybe to delete low temp protection funciton isn't a good idea, but at what degree does temperature protection start? for sure, you already did a lot for us, maybe someone can try to calibrate.

tinfever, 2024-02-04:
That's good to know the original had that under-temperature protection. It would make the most sense to update my firmware with a lower threshold temperature, but that will require updating the look up tables and possibly modifying my poorly implemented binary search algorithm. I won't be able to do that anytime soon unfortunately.