tinfever/FW-Dyson-BMS #42: Regenerate firmware with temperature annotations removed

liujixu1984, 2024-02-03:
I have disabled the temperature protection function myself and generated a new firmware here for everyone who needs to download and use. Please remove the txt suffix after downloading
[注释掉温度，320A，2.7V截止.hex.txt](https://github.com/tinfever/FU-Dyson-BMS/files/14150875/320A.2.7V.hex.txt)

Lost73, 2024-03-13:
Thanks a lot for sharing, i was planning to do same as here in caribe with high temp, it's impossible to charge without disconnecting and reset temp error. Didn't you find a way to increase the temp setting ? Is chip temp is hard set ?

xing0129, 2024-07-31:
> Thanks a lot for sharing, i was planning to do same as here in caribe with high temp, it's impossible to charge without disconnecting and reset temp error. Didn't you find a way to increase the temp setting ? Is chip temp is hard set ?

You can find the temp limit in the file config.h

dvdiamond, 2024-08-07:
> I have raised the temperature from 50 to 60 and 60 to 65 in the file configure.h. Now, it works well. Before , 12 red light flash

Can you share the hex firmware?

azhe, 2024-08-29:
> > I have disabled the temperature protection function myself and generated a new firmware here for everyone who needs to download and use. Please remove the txt suffix after downloading
> > [注释掉温度，320A，2.7V截止.hex.txt](https://github.com/tinfever/FU-Dyson-BMS/files/14150875/320A.2.7V.hex.txt)
> 
> I have raised the temperature from 50 to 60 and 60 to 65 in the file configure.h. Now, it works well. Before , 12 red light flash

Could you kindly share the HEX file? I would really appreciate it.