tinfever/FW-Dyson-BMS #43: Can't be fully charged if only check voltage

turenabo, 2024-02-04:
Hi Tinfever, I learned a bit about lithium batteries today. The battery can't be fully charged if we stop charge when the voltage just reachs 4.2, attached VTC4 charge curve. How do you think about it?
[image]

tinfever, 2024-02-04:
You're correct. Ideally you would finish the charging cycle with a constant voltage phase but the Dyson wall chargers don't support that. The charger algorithm I implemented, stopping when the voltage hits 4.2, waiting for a time, and then charging again until the voltage hits 4.2, and repeating the process until the time to hit 4.2V is short, is exactly how the stock battery firmware handles charging. I'm guessing Dyson determined that the benefit if that extra few percent of capacity wasn't worth the cost if implementing a charger with constant voltage capability. I can't really blame them for that.

It's also possible that a multi-cell battery pack like this is harder to implement constant voltage charging for, since the cells won't be perfectly balanced and so you probably can't just use the whole pack voltage of 4.2V x 6 cells = 25.2V for the constant voltage level.

turenabo, 2024-02-05:
> You're correct. Ideally you would finish the charging cycle with a constant voltage phase but the Dyson wall chargers don't support that. The charger algorithm I implemented, stopping when the [...]

I think the 2nd reason is more reasonable, and I guess by same reason telsla's battery pack has balance funtion for each cell. The 700V solution currently used in electric vehicles means that more costs must be spent on battery balancing.