tinfever/FW-Dyson-BMS #47: V8 SV10 PCB pics

npaisnel, 2024-04-01:
As per mention in the text, here are some pics of PCB from SV10

[
[image]
[image]
[image]
[image]
[image]
](url)

npaisnel, 2024-04-01:
[image]
[image]
[image]
[image]
[image]
[image]

Fockewulf44, 2024-12-06:
Do you have pinout for N1, N3, N4, N8, N9?