tinfever/FW-Dyson-BMS #49: 12 blue flashes and 70 yellow flashes

Garovich, 2024-04-03:
First of all, thank you for your titanic work on this bms firmware.
But please help me if possible.
The situation is as follows.
The firmware loaded without problems, but after a short time of use, the controller refused to work.
when pressing the button on the battery, the blue indicator lights 12 times, 4 times 3 flashes, after that the green indicator lights up and goes out.
At first, when connecting the charger, the orange indicator flashed 6 times, the battery was charging a little and that was it.
When the charging device was subsequently connected, the orange flashes only increased and reached 70 flashes, and the battery itself does not charge.
After that, I started measuring the voltage on each battery and one of them was lower than all the others, 3.8 volts and the other 4.1.
Then I replaced this one battery with the same one (from another battery), charged each battery to 4.18 volts and equalized the voltage.
Then I flashed the controller again with your firmware, but the problem with 12 blue flashes and 70 orange flashes during charging did not disappear.
I have already cleaned the memory of the controller through Pickit 3.0, but it does not give any result.
Please help if you have the opportunity.
Maybe I'm doing something wrong...

I'm not a techie, I just followed your guide
thank you

tinfever, 2024-04-04:
Hello, The LED indicator behavior you mention is bizarre. It sounds like when you press the button on the battery, it is immediately  asserting the low battery code, which is 3 blue flashes, and all error codes are repeated three times. However, you says the 3 blue flash code showed four times, which I can't explain. Normally, after the error codes have finished showing, it would go to the idle state, where you would see the green LED, and then after 30 seconds it should go to sleep and the LED should turn off.

I'm wondering if there is a hardware issue with your BMS board. If it is unable to read the voltage from one of the cells, it might read that cell voltage as 0V, which would explain the 70 yellow flash imbalance code (indicating a 3.5V imbalance) when you connect the charger.

This reminds me of someone who has a similar issue: https://github.com/tinfever/FU-Dyson-BMS/issues/38

If you could record a video showing these error codes, that might be helpful to understand more, but I don't think that will change the outcome of your board. Unless you want to dig deeper into the hardware and try to find the issue and repair it, which I can't really help with remotely, you may be best suited by replacing the BMS board with an aftermarket one you can find on eBay or Aliexpress. That will still require soldering to remove the BMS board though.

Garovich, 2024-04-04:
Hello.  
Thanks for the link to issue #39.  
I looked at the board, but I didn't see any deformations.  Resistor R24 is intact, like all the others. 

Unfortunately, I can't record the video, because I disassembled the battery again, I want to try to put the batteries on another board that was left by the donor, there is still the original firmware.  Now I charge each element separately, and also check the capacity of each element.  However, I cannot check the resistance of the elements at home, since I do not have such a tool.  
Regarding the blue flashes.  
At first when I first downloaded your firmware everything worked fine, but the Dyson only worked for a couple of minutes and showed an error of three blue flashes, then the green one.  When the battery went into sleep mode, I was able to operate the vacuum cleaner again for about one minute.  And then it showed three blue flashes again.  Next I connected the charger to the battery.  First it showed six orange flashes and then the battery was charging.  After several times of such charging of the battery and operation of the vacuum cleaner, an error began to appear when pressing the trigger, three blue flashes, which were repeated 4 times.  After that, a green flash was lit and the battery went into sleep mode.  Later, when the charger was connected, it started to emit 30 (but this is not exact) orange flashes, and it is possible that the number of orange flashes increased each time.  (I could be wrong).  After the orange flashes, a green flash came on and that was it.  That is, the battery did not charge at all.  Yesterday I measured the total voltage of the battery without the BMS board, and it showed about 25.2 volts.  However, after pressing the button, the same error flashed, 4 times 3 blue flashes and then one green.  

Now I will try with another board, then I will write if the result was positive.  Maybe some element is not working.

Garovich, 2024-04-05:
in a word, I still did it!))).  
on another board everything worked without any problems.  
My Dyson is working, even better than it was before the firmware.  
One orange light is on during charging and everything is fine.  
As for the old board, maybe some resistor burned out or something.  
Thank you again for your work and tips.  It was a very good experience for me.

tinfever, 2024-04-08:
Great! I'm glad it is working now. Thanks for reporting back.

closed 2024-04-08 by tinfever