tinfever/FW-Dyson-BMS #50: Is it possible to use rasberry PI or arduino to program the firmware?

JacksonC2024, 2024-04-14:
Sorry if this has been covered before, but I just wonder if this can be done via other hardware? 

I already have a rasberry Pi and Arduino and wonder if they can be used instead of the PICKIT.

JacksonC2024, 2024-04-15:
Some research shows that maybe this can be used. I am going to give it a try. 

https://github.com/jaromir-sukuba/a-p-prog

protocold, 2024-05-13:
it didnt work.. so i bought a pickit and it is now done.

Nickduino, 2024-05-27:
> I am going to give it a try.

Keep us posted!

kumdzio, 2025-01-02:
Unfortunately that programmer is not supporting our chip which is PIC16LF1847.
Additionally our chip is locked to be programmed only using high voltage programming.
I am currently working on software which will enable arduino to program the battery but progress is slow. If I ever finish the project for sure i will upload all details here of even maybe ask to add note about it into readme in this repo.

Nickduino, 2025-01-08:
> Edit: i have successfully flashed hex from release using my programmer.
> You can find it here: https://github.com/kumdzio/PIC16LF1847arduino-programmer

Seems great!