tinfever/FW-Dyson-BMS #51: V6 PCB: 233718 FU-Dyson-BMS compatible?

Boysolder, 2024-06-05:
Greetings enthusiasts, great inspirational read for a novice like myself...,well outside my comfort zone and loving it. V6 battery on last legs, but still working albeit briefly, looking at upgrading stock VTC4 cells with HJ2, and if that goes well will update the firmware. Case opened successfully, removed 1 cell and PCB 233718 "not listed" (looks very similar to 188002) or SVxx! Pictures attached, if BMS firmware is compatible will grab a PICKit 3.5 and dive in head first. Thank you in advance. Cheers Boysolder![image]

tinfever, 2024-06-05:
I'd be surprised if that board didn't work with the firmware but I can't guarantee it. If the board is locked out, you might want to first try unlocking it while keeping the factory firmware by editing specific values in the EEPROM. Someone else discovered this procedure and there is more info here: https://www.eevblog.com/forum/reviews/dyson-v7-trigger-cordless-vacuum-teardown-of-battery-pack/msg4028665/#msg4028665

Boysolder, 2024-06-16:
Hi Tinfever, happy to report your BMS firmware worked on my mentioned board. Initially it was bricked during cell replacement and my eeprom codes were different to those provided by other member and didn't work for me with the factory firmware. So I installed your firmware and it's now all good, Initially had red fault codes (brown out) upon installing battery back on vacuum. Required reconnection to PICkit to clear. Needed to do this twice but now all good. Thanks again for your information. Hope my feedback helps others. Cheers Solderboy "FU Dyson"!