tinfever/FW-Dyson-BMS #53: Where Google Sheet About Related Thermistor?

KerisKR, 2024-06-21:
Where Google Sheet About Related Thermistor?
hello i use B value 4200 NTC Thermistor in V6 BMS 
Dafault B value 3500 (Unpack BMS From Battery assy  To Dafault  Thermistor Broken)

Because of the difference between 3500 and 4200
Regarding temperature, I think I need to find the array value of the temperature part.
I would like to obtain a link to the Google sheet written in Thermistor.h
"/* First value in array is voltage at a given temperature, in 10mV steps. Ex. value of 24 = 240mV
 Second value is temperature in Celsius
 Data was computed in Google Sheets with the following layout and formulas:"
In thermistor.H Comment

Please google sheet link give me

qipsys, 2024-09-06:
in the file thermistor.H there are sufficient instructions on how to create your own table, for example in Excel with cell addresses and all formulas

tinfever, 2024-09-14:
Sorry for the delay. I've attached the spreadsheet I did the calculations in. Once the values were calculated I think I copied out the temperature and divided/rounded voltage values to a new sheet, swapped the order of the columns, sorted the data from lowest to highest voltage, and used a random online tool to convert the data into a C array format. Hope this helps.
[Thermistor Curve Calculator.xlsx](https://github.com/user-attachments/files/17003570/Thermistor.Curve.Calculator.xlsx)

closed 2024-09-14 by tinfever