tinfever/FW-Dyson-BMS #68: Need bootloader for new PIC16LF1847?

mtle5554, 2025-03-27:
Hello,
Im very new, My V6 bms was replaced Pic16lf1847 microcontroller chip by new one. Question is does it need a bootloader before it can work properly and how can i find and install its bootloader for it please? Sorry for my Eng. Thank you

tinfever, 2025-03-31:
Hello,
The PIC16LF1847 does not have a software bootloader in this case. It has the firmware programmed directly using a PICKit.

mtle5554, 2025-03-31:
> Hello, The PIC16LF1847 does not have a software bootloader in this case. It has the firmware programmed directly using a PICKit.

Thank you very much for your reply. After new PIC replacement and installed fw by pickit, everything works fine in the first time but after leds off board can not start again (seems to fall into deep sleep) untill I short Vpp to Gnd or cut off B- and connect again to wake it up. I have also replaced 1 more PIC and ISL94208 but nothing change. I have checked wake up line ok. I found MCP1525 2.5v reference died and replaced by resisters as voltage divider = 2.47v, do you think it is good?