tinfever/FW-Dyson-BMS #72: SV23 GEN5DT PCB:537922

Jesusrgb, 2025-05-23:
Is there any reason why we shouldn't try to install the custom firmware or unlock the device?

[image]
[image]

93510F96
2078.1J
2245B3

emrecatal00-ship-it, 2026-08-28:
Hello, I tried connecting via J-Link but the connection couldn't be established. Could you please provide the connection details on the BMS?

serhiy1978, 2026-08-29:
[image]

serhiy1978, 2026-08-29:
I think so; external power might be needed.

emrecatal00-ship-it, 2026-08-29:
Is the processor code Nuvoton M031SIAAE or ATSAMB11G18A?

serhiy1978, 2026-08-29:
I’m not exactly sure which one it is; I selected and tested them to see which ones would connect.

atmu5747-sketch, 2026-10-20:
Merhaba bulabildinmi

sekoban1, 2026-10-27:
Yes, I figured out exactly what this chip is. I managed to connect to it using Segger J-Link. I even managed to read the hex codes from the dead BMS board. However, I couldn't bring the BMS board back to life. Maybe a more experienced colleague can solve this.

sekoban1, 2026-10-27:
As you know, some Dyson V15 and Gen5 BMS boards don't have Atmel chips. While reading comments in some threads, I know many people couldn't connect to this board. After much effort, I identified the chip and managed to connect using J-Link. The chip is an O2 Micro Cortex-MO, and its full name is 93510F96LN. I managed to read the hex code from a dead BMS board. The CPU connection via SWD works perfectly. A repairman I know revived the board without installing new software, but he's not telling anyone how he did it. Perhaps by sharing information with more experienced people, we can overcome this BMS board issue and everyone can benefit. I'm ready to provide more details if you want.

atmu5747-sketch, 2026-10-28:
Anladım keşke nasil yaptığını ogrenebilsek onunla nasıl iletişime gecebilirim

dackerman38, 2026-11-02:
> As you know, some Dyson V15 and Gen5 BMS boards don't have Atmel chips. While reading comments in some threads, I know many people couldn't connect to this board. After much effort, I identified the [...]

Could you share the hex code you read from your board ?
As for the repairman you know, maybe he changed the bms or the bms wasnt blocked for some unknown reason?!!

serhiy1978, 2026-11-03:
From which board did you read the dump? gen5 or V15 ?

serhiy1978, 2026-11-03:
Connecting ...
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: J-Link V12 compiled Jun 25 2026 10:23:52
 - Probe/ Programmer S/N: 603000869
 - Device "OZ93510F96LN" selected.
 - Target interface speed: 4000 kHz (Fixed)
 - VTarget = 3.320V
 - Found SW-DP with ID 0x0BB11477
 - DPIDR: 0x0BB11477
 - CoreSight SoC-400 or earlier
 - Scanning AP map to find all available APs
 - AP[1]: Stopped AP scan as end of AP map has been reached
 - AP[0]: AHB-AP (IDR: 0x04770021, ADDR: 0x00000000)
 - Iterating through AP map to find AHB-AP to use
 - AP[0]: Core found
 - AP[0]: AHB-AP ROM base: 0xE00FF000
 - CPUID register: 0x410CC200. Implementer code: 0x41 (ARM)
 - Found Cortex-M0 r0p0, Little endian.
 - FPUnit: 4 code (BP) slots and 0 literal slots
 - CoreSight components:
 - ROMTbl[0] @ E00FF000
 - [0][0]: E000E000 CID B105E00D PID 000BB008 SCS
 - [0][1]: E0001000 CID B105E00D PID 000BB00A DWT
 - [0][2]: E0002000 CID B105E00D PID 000BB00B FPB
 - Executing init sequence ...
  - Initialized successfully
 - Target interface speed: 4000 kHz (Fixed)
 - Found 1 JTAG device. Core ID: 0x0BB11477 (None)
 - Connected successfully
Reading entire flash chip ...
 - 35 sectors, 4 ranges, 0x10000 - 0x17FFF, 0x18040 - 0x1807F, 0x19000 - 0x190FF, 0x60000000 - 0x6000FFFF
 - Start of determining flash info (Bank 0 @ 0x00010000)
 - End of determining flash info
 - Flash bank info:
 - 32 * 1 KB @ 0x00010000
 - Start of preparing flash programming
 - End of preparing flash programming
 - Blank checking 0x00010000 - 0x00017FFF
 - Start of restoring
 - End of restoring
 - Start of determining flash info (Bank 1 @ 0x00018040)
 - End of determining flash info
 - Flash bank info:
 - 1 * 64 bytes @ 0x00018040
 - Start of preparing flash programming
 - End of preparing flash programming
 - Blank checking 0x00018040 - 0x0001807F
 - Start of restoring
 - End of restoring
 - Start of determining flash info (Bank 2 @ 0x00019000)
 - End of determining flash info
 - Flash bank info:
 - 1 * 256 bytes @ 0x00019000
 - Start of preparing flash programming
 - End of preparing flash programming
 - Blank checking 0x00019000 - 0x000190FF
 - Start of restoring
 - End of restoring
 - WARNING: Flash bank 3 disabled, skipped.
 - Target memory read successfully. (33088 bytes, 3 ranges) - Completed after 0.165 sec
Saving data file [D:\v15 oz93510f96.mot] ...
 - Data file saved successfully (33088 bytes, 3 ranges)
Saving data file [D:\v15 oz93510f96.bin] ...
 - Data file saved successfully (33088 bytes, 3 ranges)
Saving data file [D:\v15 oz93510f96.hex] ...